guysoft/CustomPiOS #139: Module docker causes build failures and

davidzwa, 2021-08-05:
Built an image without docker and then I added the docker module.

**CustomPiOS:** devel
**Error summary:** cargo or rust cause wheel build which never finishes. The fact that a wheel build is necessary tells me something is wrong with the pip packages to start with. I could share the errors TL;DR a docker-compose dependency
**Error log:** please visit my public repository and this build log line https://github.com/davidzwa/mtb3dpi-image/pull/1/checks?check_run_id=3255944271#step:8:1235 
**Attempt to fix**: I ran the instructions on https://dev.to/elalemanyo/how-to-install-docker-and-docker-compose-on-raspberry-pi-1mo as Ive done many times before (hence this custom pi OS attempt) with success. The instructions install much less deps, which makes me wonder if we can simplify it here as well. 
**Reproduction:**
1) initiate a new custom pi OS using Docker on Windows 10 and WSL or use a Github Actions workflow just like FarmPi (https://github.com/mkevenaar/FarmPi/blob/main/.github/workflows/build.yaml)
2) download custom OS like raspios-buster-lite or raspbian (default)
3) adjust src/config with 
```
export DIST_NAME=MTB3DPI
export DIST_VERSION=0.1.0

export BASE_USER=pi
export BASE_USER_PASSWORD=raspberry
export BASE_IMAGE_ENLARGEROOT=500
export BASE_IMAGE_RESIZEROOT=250

export MODULES="base(network, docker, mtb3dpi, password-for-sudo)"
```
4) build

I hope this information is dense enough. Willing to PR.

guysoft, 2021-08-05:
Strange, it does finish on my builds, just takes a while. About 15 mins on my Pi4. Might be that github are emulating arm and therefore it would take longer, about half an hour.

Here is an example log from MtigOS that uses the docker module:
```
17:52:49   Created wheel for cffi: filename=cffi-1.14.6-cp37-cp37m-linux_aarch64.whl size=346183 sha256=61cc4bd42b98fb68cf0fad012ad13167bad847a0f3a3d117be597400cf8440d0
17:52:49   Stored in directory: /root/.cache/pip/wheels/44/5d/fc/189fa782e27940a19fa887d7896073a3691422f61b85f41a17
17:52:49   Building wheel for cryptography (PEP 517): started
17:53:49   Building wheel for cryptography (PEP 517): still running...
17:56:50   Building wheel for cryptography (PEP 517): still running...
17:57:50   Building wheel for cryptography (PEP 517): still running...
17:59:32   Building wheel for cryptography (PEP 517): still running...
17:59:35   Building wheel for cryptography (PEP 517): finished with status 'done'
17:59:35   Created wheel for cryptography: filename=cryptography-3.4.7-cp37-cp37m-linux_aarch64.whl size=1464730 sha256=043121d1b8caaef099e8581a0ad7ebadafd770a5df8664c001415facd6dcf259
17:59:35   Stored in directory: /root/.cache/pip/wheels/0f/d4/8b/207901c6ba23c69e3823daeebe87ffc684f28662da0c29bb39
17:59:35   Building wheel for pynacl (PEP 517): started
18:00:40   Building wheel for pynacl (PEP 517): still running...
18:01:42   Building wheel for pynacl (PEP 517): still running...
18:02:42   Building wheel for pynacl (PEP 517): still running...
18:03:46   Building wheel for pynacl (PEP 517): still running...
18:04:47   Building wheel for pynacl (PEP 517): still running...
18:05:50   Building wheel for pynacl (PEP 517): still running...
18:06:54   Building wheel for pynacl (PEP 517): still running...
18:08:02   Building wheel for pynacl (PEP 517): still running...
18:09:02   Building wheel for pynacl (PEP 517): still running...
18:10:04   Building wheel for pynacl (PEP 517): still running...
18:11:10   Building wheel for pynacl (PEP 517): still running...
18:12:14   Building wheel for pynacl (PEP 517): still running...
18:13:18   Building wheel for pynacl (PEP 517): still running...
18:14:19   Building wheel for pynacl (PEP 517): still running...
18:15:15   Building wheel for pynacl (PEP 517): finished with status 'done'
18:15:15   Created wheel for pynacl: filename=PyNaCl-1.4.0-cp37-cp37m-linux_aarch64.whl size=605424 sha256=5b81bb6b636aa2a6ae338f4639b4a0041980b3e394d8d6d812bb4e4fe06c38f6
18:15:15   Stored in directory: /root/.cache/pip/wheels/f9/fb/34/d15387bc2e4a748e0eb2b9788c34e1d0c2167265e04bc6bb49
18:15:15   Building wheel for pyrsistent (PEP 517): started
18:15:18   Building wheel for pyrsistent (PEP 517): finished with status 'done'
18:15:18   Created wheel for pyrsistent: filename=pyrsistent-0.18.0-cp37-cp37m-linux_aarch64.whl size=100364 sha256=d1849500d8716a9ed04eaa8302d262b3bd672032728ad165b4ef0e05c02eff85
18:15:18   Stored in directory: /root/.cache/pip/wheels/54/35/6e/9827c774ec40ae001ae2a2ef063268add65762c4014951e97c
18:15:18 Successfully built dockerpty docopt bcrypt cffi cryptography pynacl pyrsistent
18:15:19 Installing collected packages: pycparser, cffi, zipp, websocket-client, typing-extensions, pynacl, cryptography, bcrypt, pyrsistent, paramiko, importlib-metadata, docker, attrs, texttable, python-dotenv, jsonschema, docopt, dockerpty, distro, cached-property, docker-compose
18:15:22 Successfully installed attrs-21.2.0 bcrypt-3.2.0 cached-property-1.5.2 cffi-1.14.6 cryptography-3.4.7 distro-1.6.0 docker-5.0.0 docker-compose-1.29.2 dockerpty-0.4.1 docopt-0.6.2 importlib-metadata-4.6.3 jsonschema-3.2.0 paramiko-2.7.2 pycparser-2.20 pynacl-1.4.0 pyrsistent-0.18.0 python-dotenv-0.19.0 texttable-1.6.4 typing-extensions-3.10.0.0 websocket-client-0.59.0 zipp-3.5.0
```

Also attaching full log for reference.
[consoleText.txt](https://github.com/guysoft/CustomPiOS/files/6942093/consoleText.txt)

A way to speed it up would be to build the wheel in advance, or use a 64bit build since they 

Also there is this comment with a fix I don't fully understand:
https://github.com/pyca/pynacl/issues/553#issuecomment-582506232

davidzwa, 2021-08-06:
To re-iterate: with the installation steps I linked above a wheel build is not needed at all.

https://dev.to/elalemanyo/how-to-install-docker-and-docker-compose-on-raspberry-pi-1mo

guysoft, 2021-08-06:
Ok, gotcha.

Is the apt upgrade required? Any idea what it is fixing?

Apart from that, the module also runs the same up to here
```
curl -sSL https://get.docker.com | sh
sudo usermod -aG docker ${USER}
```

The lines here in your solution:
```
sudo apt-get install libffi-dev libssl-dev
sudo apt install python3-dev
sudo apt-get install -y python3 python3-pip

sudo pip3 install docker-compose
```

Are replaced by this in the docker module ([source](https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/docker/start_chroot_script#L30)):
```
    apt-get install -y python3 python3-distutils python3-dev python3-testresources gcc libffi-dev build-essential libssl-dev cargo
    # Docker-compose
    wget https://bootstrap.pypa.io/get-pip.py -O - | python3
    pip3 install --ignore-installed PyYAML
    pip3 install docker-compose
```

I rather not change the base image as much as I can, since we expect Rpi-Foundation to maintain their image. Though we could add a flag for now ``DOCKER_WITH_UPGRADE=y``. That when used would use the apt-get upgrade script you suggested.

At least until Rpi upgrade to Bullseye (which would be likely before 2022 when Buster reaches EOL).

davidzwa, 2021-08-07:
Yeah it would be bad if we'd require `apt upgrade`. Lets aim to do without. I'll try that now using Raspbian (default).

davidzwa, 2021-08-07:
I was able to build the raspbian base without problems. Here's the start script:
```
#!/usr/bin/env bash
# <Script Name>
# <Description what this module does>
# Written by <Author>
# GPL V3
########

# Source error handling, leave this in place
set -x
set -e

source /common.sh
install_cleanup_trap

unpack /filesystem/home/pi /home/"${BASE_USER}" "${BASE_USER}"
unpack /filesystem/home/root /root root
unpack /filesystem/boot /boot

echo "This is an example module, install and set up what you need here!"
echo "$MTB3DPI_VAR"

sudo apt-get update

curl -sSL https://get.docker.com | sh
sudo usermod -aG docker ${BASE_USER}
sudo apt -y install libffi-dev libssl-dev
sudo apt -y install python3-dev
sudo apt -y install python3 python3-pip
sudo pip3 install docker-compose
sudo systemctl enable docker

# Unpack root at the end, so files are modified before
unpack /filesystem/root /
```

Havent tested docker-compose on the image + host, but I trust it will succeed

guysoft, 2021-08-09:
Looks like what you change is installing python3-pip from the distribution and not for the internet?
I can fix that

davidzwa, 2021-08-09:
Looking at the `devel` docker module, it definitely looks like it.

```
    wget https://bootstrap.pypa.io/get-pip.py -O - | python3
    pip3 install --ignore-installed PyYAML
    pip3 install docker-compose
```

By looking at the get-pip file it downloads pip3 `21.2.3` which is the latest version. Personally I do not like that as the CustomPiOS build for an image might suddenly break with newer pip versions. Would be great if we fixated the version (or left it to the developers to upgrade it).
https://packages.debian.org/buster/python3-pip
Version 18.1-5 will be included if we depend on Debian buster

guysoft, 2021-08-11:
Ok, tried to add python3-pip and it still takes a long time, log:
```
23:29:29 + apt-get install -y python3 python3-distutils python3-dev python3-testresources gcc libffi-dev build-essential libssl-dev cargo python3-pip
23:29:29 Reading package lists...
23:29:29 Building dependency tree...
23:29:29 Reading state information...
23:29:30 build-essential is already the newest version (12.6).
23:29:30 gcc is already the newest version (4:8.3.0-1+rpi2).
23:29:30 gcc set to manually installed.
23:29:30 python3 is already the newest version (3.7.3-1).
23:29:30 python3 set to manually installed.
23:29:30 The following additional packages will be installed:
23:29:30   dh-python gir1.2-glib-2.0 libcurl3-gnutls libexpat1-dev
23:29:30   libgirepository-1.0-1 libhttp-parser2.8 libllvm7 libpython3-dev
23:29:30   libpython3.7-dev libstd-rust-1.41 libstd-rust-dev python-pip-whl
23:29:30   python3-asn1crypto python3-cffi-backend python3-crypto python3-cryptography
23:29:30   python3-dbus python3-entrypoints python3-gi python3-keyring
23:29:30   python3-keyrings.alt python3-lib2to3 python3-linecache2 python3-pbr
23:29:30   python3-secretstorage python3-setuptools python3-traceback2
23:29:30   python3-unittest2 python3-wheel python3-xdg python3.7-dev rust-gdb rustc
23:29:30 Suggested packages:
23:29:30   cargo-doc libssl-doc python-crypto-doc python-cryptography-doc
23:29:30   python3-cryptography-vectors python-dbus-doc python3-dbus-dbg gnome-keyring
23:29:30   libkf5wallet-bin gir1.2-gnomekeyring-1.0 python-secretstorage-doc
23:29:30   python-setuptools-doc gdb-doc rust-doc rust-src lld-7
23:29:30 The following NEW packages will be installed:
23:29:30   cargo dh-python gir1.2-glib-2.0 libcurl3-gnutls libexpat1-dev libffi-dev
23:29:30   libgirepository-1.0-1 libhttp-parser2.8 libllvm7 libpython3-dev
23:29:30   libpython3.7-dev libssl-dev libstd-rust-1.41 libstd-rust-dev python-pip-whl
23:29:30   python3-asn1crypto python3-cffi-backend python3-crypto python3-cryptography
23:29:30   python3-dbus python3-dev python3-distutils python3-entrypoints python3-gi
23:29:30   python3-keyring python3-keyrings.alt python3-lib2to3 python3-linecache2
23:29:30   python3-pbr python3-pip python3-secretstorage python3-setuptools
23:29:30   python3-testresources python3-traceback2 python3-unittest2 python3-wheel
23:29:30   python3-xdg python3.7-dev rust-gdb rustc
23:29:30 0 upgraded, 40 newly installed, 0 to remove and 38 not upgraded.
23:29:30 Need to get 105 MB of archives.
23:29:30 After this operation, 278 MB of additional disk space will be used.
23:29:30 Get:1 http://archive.raspberrypi.org/debian buster/main armhf libssl-dev armhf 1.1.1d-0+deb10u6+rpt1 [1584 kB]
23:29:31 Get:5 http://archive.raspberrypi.org/debian buster/main armhf python-pip-whl all 18.1-5+rpt1 [1591 kB]
23:29:31 Get:6 http://archive.raspberrypi.org/debian buster/main armhf python3-pip all 18.1-5+rpt1 [171 kB]
23:29:31 Get:2 http://mirror.ossplanet.net/raspbian/raspbian buster/main armhf libcurl3-gnutls armhf 7.64.0-4+deb10u2 [292 kB]
23:29:32 Get:3 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf libhttp-parser2.8 armhf 2.8.1-1 [19.1 kB]
23:29:32 Get:11 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-lib2to3 all 3.7.3-1 [76.7 kB]
23:29:33 Get:12 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-distutils all 3.7.3-1 [142 kB]
23:29:33 Get:4 http://mirror.ossplanet.net/raspbian/raspbian buster/main armhf libllvm7 armhf 1:7.0.1-8+rpi3+deb10u2 [11.4 MB]
23:29:33 Get:13 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf dh-python all 3.20190308 [99.3 kB]
23:29:34 Get:14 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf libgirepository-1.0-1 armhf 1.58.3-2 [78.8 kB]
23:29:34 Get:15 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf gir1.2-glib-2.0 armhf 1.58.3-2 [143 kB]
23:29:34 Get:17 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf libffi-dev armhf 3.2.1-9 [159 kB]
23:29:34 Get:19 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf libpython3-dev armhf 3.7.3-1 [20.1 kB]
23:29:35 Get:20 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-asn1crypto all 0.24.0-1 [78.2 kB]
23:29:35 Get:21 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-cffi-backend armhf 1.12.2-1 [70.1 kB]
23:29:36 Get:24 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-dbus armhf 1.2.8-3 [95.9 kB]
23:29:36 Get:26 http://raspbian.raspberrypi.org/raspbian buster/main armhf python3-dev armhf 3.7.3-1 [1264 B]
23:29:37 Get:27 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-entrypoints all 0.3-1 [5508 B]
23:29:37 Get:28 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-gi armhf 3.30.4-1 [161 kB]
23:29:38 Get:29 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-secretstorage all 2.3.1-2 [14.2 kB]
23:29:38 Get:30 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-keyring all 17.1.1-1 [43.1 kB]
23:29:38 Get:31 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-keyrings.alt all 3.1.1-1 [18.2 kB]
23:29:38 Get:32 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-linecache2 all 1.0.0-3 [12.4 kB]
23:29:38 Get:33 http://mirror.isoc.org.il/pub/raspbian/raspbian buster/main armhf python3-setuptools all 40.8.0-1 [306 kB]
[... cut ...]
23:48:59   Running setup.py bdist_wheel for pynacl: still running...
23:50:00   Running setup.py bdist_wheel for pynacl: still running...
23:51:06   Running setup.py bdist_wheel for pynacl: still running...
23:52:06   Running setup.py bdist_wheel for pynacl: still running...
23:52:43   Running setup.py bdist_wheel for pynacl: finished with status 'done'
23:52:43   Stored in directory: /root/.cache/pip/wheels/c2/d9/ca/81e002fb0cee770c198715dace9edab919daba3ae49bd287a0
23:52:44 Successfully built pyrsistent bcrypt pynacl
```

davidzwa, 2021-08-12:
```
apt-get install -y python3 python3-distutils python3-dev python3-testresources gcc libffi-dev build-essential libssl-dev cargo python3-pip
```

There's a lot more in that install than what I am using.
```
sudo apt-get install libffi-dev libssl-dev
sudo apt install python3-dev
sudo apt-get install -y python3 python3-pip
```

The package difference:
- build-essential (??)
- python3-distuils (??)
- python3-testresources (??)
- gcc (???)
- cargo

I highly suggest removing these build tool related packages like gcc and cargo (rust?) what you dont need.

davidzwa, 2021-08-12:
Here's my log when building for raspbian

```
+ echo 'This is a module variable'
This is a module variable
+ sudo apt-get update
sudo: unable to resolve host 362f98fef28d: Name or service not known
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [378 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]
Get:6 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Fetched 13.6 MB in 7s (1985 kB/s)
Reading package lists...
+ curl -sSL https://get.docker.com
+ sh
# Executing docker install script, commit: 0e685c6ac0bddd7b2ba7bcaaeb519746ad249a29

WSL DETECTED: We recommend using Docker Desktop for Windows.
Please get Docker Desktop from https://www.docker.com/products/docker-desktop

You may press Ctrl+C now to abort this script.
+ sleep 20
+ sh -c apt-get update -qq >/dev/null
+ sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
+ sh -c curl -fsSL "https://download.docker.com/linux/raspbian/gpg" | gpg --dearmor -o /usr/share/keyrings/docker-archive-keyring.gpg
+ sh -c echo "deb [arch=armhf signed-by=/usr/share/keyrings/docker-archive-keyring.gpg] https://download.docker.com/linux/raspbian buster stable" > /etc/apt/sources.list.d/docker.list
+ sh -c apt-get update -qq >/dev/null
+ sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq --no-install-recommends  docker-ce-cli docker-ce >/dev/null
+ version_gte 20.10
+ [ -z  ]
+ return 0
+ sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq docker-ce-rootless-extras >/dev/null

================================================================================

To run Docker as a non-privileged user, consider setting up the
Docker daemon in rootless mode for your user:

    dockerd-rootless-setuptool.sh install

Visit https://docs.docker.com/go/rootless/ to learn about rootless mode.

To run the Docker daemon as a fully privileged service, but granting non-root
users access, refer to https://docs.docker.com/go/daemon-access/

WARNING: Access to the remote API on a privileged Docker daemon is equivalent
         to root access on the host. Refer to the 'Docker daemon attack surface'
         documentation for details: https://docs.docker.com/go/attack-surface/

================================================================================

+ sudo usermod -aG docker pi
sudo: unable to resolve host 362f98fef28d: Name or service not known
+ sudo apt -y install libffi-dev libssl-dev
sudo: unable to resolve host 362f98fef28d: Name or service not known

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libssl1.1
Suggested packages:
  libssl-doc
The following NEW packages will be installed:
  libffi-dev libssl-dev
The following packages will be upgraded:
[... 325 lines cut ...]
Collecting pyrsistent>=0.14.0 (from jsonschema<4,>=2.5.1->docker-compose)
  Downloading https://www.piwheels.org/simple/pyrsistent/pyrsistent-0.18.0-cp37-cp37m-linux_armv7l.whl (100kB)
Collecting attrs>=17.4.0 (from jsonschema<4,>=2.5.1->docker-compose)
  Downloading https://files.pythonhosted.org/packages/20/a9/ba6f1cd1a1517ff022b35acd6a7e4246371dfab08b8e42b829b6d07913cc/attrs-21.2.0-py2.py3-none-any.whl (53kB)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from jsonschema<4,>=2.5.1->docker-compose) (40.8.0)
Collecting bcrypt>=3.1.3 (from paramiko>=2.4.2; extra == "ssh"->docker[ssh]>=5->docker-compose)
  Downloading https://www.piwheels.org/simple/bcrypt/bcrypt-3.2.0-cp37-cp37m-linux_armv7l.whl (57kB)
Collecting pynacl>=1.0.1 (from paramiko>=2.4.2; extra == "ssh"->docker[ssh]>=5->docker-compose)
  Downloading https://www.piwheels.org/simple/pynacl/PyNaCl-1.4.0-cp37-cp37m-linux_armv7l.whl (605kB)
Requirement already satisfied: cryptography>=2.5 in /usr/lib/python3/dist-packages (from paramiko>=2.4.2; extra == "ssh"->docker[ssh]>=5->docker-compose) (2.6.1)
Collecting typing-extensions>=3.6.4; python_version < "3.8" (from importlib-metadata; python_version < "3.8"->jsonschema<4,>=2.5.1->docker-compose)
  Downloading https://files.pythonhosted.org/packages/2e/35/6c4fff5ab443b57116cb1aad46421fb719bed2825664e8fe77d66d99bcbc/typing_extensions-3.10.0.0-py3-none-any.whl
Collecting zipp>=0.5 (from importlib-metadata; python_version < "3.8"->jsonschema<4,>=2.5.1->docker-compose)
  Downloading https://files.pythonhosted.org/packages/92/d9/89f433969fb8dc5b9cbdd4b4deb587720ec1aeb59a020cf15002b9593eef/zipp-3.5.0-py3-none-any.whl
Collecting cffi>=1.1 (from bcrypt>=3.1.3->paramiko>=2.4.2; extra == "ssh"->docker[ssh]>=5->docker-compose)
  Downloading https://www.piwheels.org/simple/cffi/cffi-1.14.6-cp37-cp37m-linux_armv7l.whl (346kB)
Collecting pycparser (from cffi>=1.1->bcrypt>=3.1.3->paramiko>=2.4.2; extra == "ssh"->docker[ssh]>=5->docker-compose)
  Downloading https://files.pythonhosted.org/packages/ae/e7/d9c3a176ca4b02024debf82342dab36efadfc5776f9c8db077e8f6e71821/pycparser-2.20-py2.py3-none-any.whl (112kB)
[... cut ...]
Successfully installed PyYAML-5.4.1 attrs-21.2.0 bcrypt-3.2.0 cached-property-1.5.2 cffi-1.14.6 distro-1.6.0 docker-5.0.0 docker-compose-1.29.2 dockerpty-0.4.1 docopt-0.6.2 importlib-metadata-4.6.3 jsonschema-3.2.0 paramiko-2.7.2 pycparser-2.20 pynacl-1.4.0 pyrsistent-0.18.0 python-dotenv-0.19.0 texttable-1.6.4 typing-extensions-3.10.0.0 websocket-client-0.59.0 zipp-3.5.0
+ sudo systemctl enable docker
sudo: unable to resolve host 362f98fef28d: Name or service not known
Synchronizing state of docker.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable docker
```

guysoft, 2021-08-12:
Looks like it says that PyNaCl has a wheel, but form an unexpected location:
```
Downloading https://www.piwheels.org/simple/pynacl/PyNaCl-1.4.0-cp37-cp37m-linux_armv7l.whl
```
So it its pulling it.

The url is:
https://www.piwheels.org/

It says it needs to include that in ``/etc/pip.conf`` for it to work. Not related to all the pip stuff.

This is unlike the long loop for in my builds:
```
Running setup.py bdist_wheel for pynacl: still running...
```

This actually might connect to the long standing PR #123.
Might be worth adding a module or something with better pip settings.

I does seem to appear in Rpi OS
```
cat /etc/pip.conf
[global]
extra-index-url=https://www.piwheels.org/simple
```

guysoft, 2021-08-12:
I am going to try adding:
```
pip3 install --index-url=https://www.piwheels.org/simple PyNaCl
```
And see if that makes the build finish faster, if so the full fix will be:

```bash
if [ "${BASE_DISTRO}" == "raspbian" ]; then
    pip3 install --index-url=https://www.piwheels.org/simple PyNaCl
fi
```
This way it will only do that in Rasbian

guysoft, 2021-08-15:
Think I hit something strange or a dumb misstake.
I get:
```
15:04:55 + pip3 install --index-url=https://www.piwheels.org/simple pynacl
15:04:58 Looking in indexes: https://www.piwheels.org/simple, https://www.piwheels.org/simple
15:04:58 Collecting pynacl
15:04:59   Could not find a version that satisfies the requirement pynacl (from versions: )
15:04:59 No matching distribution found for pynacl
```

davidzwa, 2021-08-15:
> I am going to try adding:
> 
> ```
> pip3 install --index-url=https://www.piwheels.org/simple PyNaCl
> ```
> 
> And see if that makes the build finish faster, if so the full fix will be:
> 
> ```shell
> if [ "${BASE_DISTRO}" == "raspbian" ]; then
>     pip3 install --index-url=https://www.piwheels.org/simple PyNaCl
> fi
> ```
> 
> This way it will only do that in Rasbian

I dont agree that only that works, as shown earlier. Shall I PR and let my github actions be proof?

guysoft, 2021-08-15:
> Error log: please visit my public repository and this build log line https://github.com/davidzwa/mtb3dpi-image/pull/1/checks?check_run_id=3255944271#step:8:1235

PyNaCl is the problem according to the log you supplied in the issue report.

Here is the output as a screenshot from that log so you dont need to scroll all the way down:
[image]

davidzwa, 2021-08-15:
No this problem is gone as I shared in this comment 😉 
https://github.com/guysoft/CustomPiOS/issues/139#issuecomment-897436023

guysoft, 2021-08-15:
I am sorry if its a little repetitive. It just also wanted to make a summery of what I understand the problem is. 
So let me try and summarize what I understood, and hopefully it could help you point me.

> No this problem is gone as I shared in this comment wink There's no wheel for pynacl in that log.
> [#139 (comment)](https://github.com/guysoft/CustomPiOS/issues/139#issuecomment-897436023)

Indeed, as I said in comment https://github.com/guysoft/CustomPiOS/issues/139#issuecomment-897537891 , it seems like the way that got solved there is because the pynacl got installed via piwheels, which is a raspberry pi-spasific wheel repository (as in not PyPi).
Here is the output from your comment https://github.com/guysoft/CustomPiOS/issues/139#issuecomment-897436023 that shows that here that pynacl is installed differently using piwheels:
```
Collecting pynacl>=1.0.1 (from paramiko>=2.4.2; extra == "ssh"->docker[ssh]>=5->docker-compose)
  Downloading https://www.piwheels.org/simple/pynacl/PyNaCl-1.4.0-cp37-cp37m-linux_armv7l.whl (605kB)
```

So that log difference from what I understood is what need to happen for the build not to break as given in the initial log of the issue.

davidzwa, 2023-04-28:
Going to close this as it might be completely different now and I'm not running docker anymore due to lingering overlay2 folders etc.
Thanks for the thorough help a while ago!

closed 2023-04-28 by davidzwa