guysoft/CustomPiOS #140: Docker image for CustomPiOS moved to github actions, but not the new verson tags

guysoft, 2021-08-15:
Hey,
So DockerHub sadly now request money for automated builds:
https://www.docker.com/blog/changes-to-docker-hub-autobuilds/

I order to workaround that I have now started building from github actions. Actions defined here:
https://github.com/guysoft/CustomPiOS/blob/devel/.github/workflows/docker-build.yml

The good news is that github actions can use docker buildx which means less dockerfile per-arch (bye bye in https://github.com/guysoft/CustomPiOS/commit/f7f3a5753aebbb883faab981af0235a0269a6544 ).

However I have yet to set up that every version tag in git will also tag a version on DockerHub. So this issue will stay open until that is fixed.
Will tag Gina when she is free.

pablosproject, 2021-08-18:
You could try using this action in your steps https://github.com/docker/metadata-action, which automatically set up labels and tags to push to docket hub.

Here is an example of how I use it, if I can do something this weekend I'll provide a PR on the action:

```
teps:
      - name: Check out the repo
        uses: actions/checkout@v2.3.4

      - name: Docker meta tag
        id: docker_meta
        uses: crazy-max/ghaction-docker-meta@v1
        with:
          images: ghcr.io/XXXXX          
          tag-sha: true

      - name: Set up Docker Buildx
        id: buildx
        uses: docker/setup-buildx-action@master

      - name: Cache Docker layers
        uses: actions/cache@v2
        with:
          path: /tmp/.buildx-cache
          key: ${{ runner.os }}-buildx-${{ github.sha }}
          restore-keys: |
            ${{ runner.os }}-buildx-
      - name: Login to GitHub Packages Docker Registry
        uses: docker/login-action@v1
        with:
          registry: ghcr.io
          username: xxxx
          password: ${{ secrets.CR_PAT }}

      - name: Push to GitHub Packages
        uses: docker/build-push-action@v2
        with:
          context: ./
          file: ./Dockerfile
          builder: ${{ steps.buildx.outputs.name }}
          push: true
          tags: ${{ steps.docker_meta.outputs.tags }}
          labels: ${{ steps.docker_meta.outputs.labels }}
          cache-from: type=local,src=/tmp/.buildx-cache
          cache-to: type=local,dest=/tmp/.buildx-cache

      - name: Logout from docker
        run: docker logout
```

pablosproject, 2021-11-05:
As the PR #144 address it shall we close this issue?

guysoft, 2021-11-05:
Yes, thanks!

closed 2021-11-05 by guysoft