guysoft/CustomPiOS #141: raspian or debian?

steviehs, 2021-08-19:
http://downloads.raspberrypi.org/raspbian_lite/images/ - this is the place, where CustomPiOS downloads the latest image - hasn't received new images since Feb. 2020. Does anybody know why? 
What about https://raspi.debian.net/? Wouldn't this be the future way to go?

cp2004, 2021-08-19:
Raspbian was renamed to Raspberry Pi OS, if you look through https://downloads.raspberrypi.org/ you will find `raspios_*` for the various different types of images

guysoft, 2021-08-19:
What @cp2004 said, he equivalent to the url you posted is now:
https://downloads.raspberrypi.org/raspios_lite_armhf/images/

steviehs, 2021-08-19:
Maybe it would be worth to change the URL to the raspios_lite images in make_custom_pi_os?
I think I will try soonish to switch our project to the debian images (though they are also not "official"), as this makes sense only on RPi4... let you know the results

guysoft, 2021-08-19:
Thanks for pointing out!
Fixed

closed 2021-08-19 by guysoft

steviehs, 2021-08-19:
nope, you should at least have tested it once :-)

steviehs, 2021-08-19:
I have added a pull request, this was an easy one.

guysoft, 2021-08-19:
Excellent, i missed the grep bit.
Merged reference #142

steviehs, 2021-08-19:
Thanks for merging fast, so my build process will work again :-)

guysoft, 2021-08-19:
BTW you might want to look in to the ubuntu 64-bit image if you are doing RaspberryPi 4-only stuff. 
might squeeze a little more performance out of it.

The make os script doesn't support it. But I maintain ubuntu to work for the OctoPi 64bit builds.

commit f9cf2840 (2021-08-19): Update to raspios_lite_armhf, fixes #141