guysoft/CustomPiOS #143: Release-info-change on a raspbian repository causes build failure

shueja-personal, 2021-08-28:
https://github.com/OctoPrint/CustoPiZer/issues/1 
This issue popped up last night on the CustoPiZer repository as I was trying to get that build method to work. When I switched to CustomPiOS and added a module that does nothing except this script: 
```#!/usr/bin/env bash

source /common.sh

sudo -u pi /home/pi/oprint/bin/pip install -U OctoPrint==1.6.1
```
I get a very similar error message to what is linked in that issue. @jneilliii found the error and fixed it in CustoPiZer (https://github.com/jneilliii/CustoPiZer/commit/11552560de50860a9a5adf35e868910bce2f9e96), and the analogous issue here is by my searching located in https://github.com/guysoft/CustomPiOS/blob/fc0cc4d2b71ceec8164af5aae72fb44b1613745b/src/modules/base/start_chroot_script#L23

I would PR it myself, but I'm not familiar enough with the wonders of docker to be able to properly test that it would fix the issue.

guysoft, 2021-08-31:
Why is the releaseinfo changeing there?
 @jneilliii any idea why that solved the issue?

jneilliii, 2021-08-31:
Something about upstream release name changes to debian. Without that flag the build fails on the first apt update.

guysoft, 2021-09-01:
What base image? I want to open an issue to https://github.com/RPi-Distro/pi-gen/issues/

shueja-personal, 2021-09-01:
I used this: https://github.com/OctoPrint/OctoPi-UpToDate/releases/tag/0.18.0.op1.6.1 with a custom module to install the custom stuff.

guysoft, 2021-09-09:
@shueja-personal That ``apt-get update`` command is enclose inside an if:

```bash
if [ "${BASE_DISTRO}" == "ubuntu" ]; then
```

So you are not supposed to reach it at all. You are setting ``${BASE_DISTRO}`` to be Ubuntu somewhere.

Make sure to set 
```
BASE_DISTRO=raspbian
```
Then that line should not run at all.

I tested here with the base image you provided, and it does not even execute that line. It runs:
```
+ install_cleanup_trap
+ set -e
+ trap cleanup SIGINT SIGTERM
+ '[' raspbian == ubuntu ']'
```
Then continues.

jneilliii, 2021-09-09:
> What base image? I want to open an issue to https://github.com/RPi-Distro/pi-gen/issues/

Whatever you are using for the last stable octopi 0.18 image, 32bit. The same error has popped up with the OctoDash install scripts, where people running the script are getting the error as well. Hope that helps. As an FYI the CustoPiZer scripts were updated and that allowed them to run again without choking on the error.

jneilliii, 2021-09-09:
> That apt-get update command is enclose inside an if

You are correct, and it another PR was sent after that corrected both apt-get commands in that file. 

https://github.com/jneilliii/CustoPiZer/commit/fc1d28021ffc93b1b8ce3252c75ed601566aa899

guysoft, 2021-09-09:
@jneilliii 
At the moment I am considering two options here.
I would very much would like to reproduce the bug here and determine what the "Something about upstream release name changes to debian" actually are. It might be something that needs to go upstream.

If not I could just modify it and put a "workaround for issue #143" comment and move on, I would just like to attempt to figure out what is going on before I do that.

jneilliii, 2021-09-09:
well, you could reproduce the error by flashing octopi 0.18 and try to install OctoDash with the command line below.

```
bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)
```

jneilliii, 2021-09-09:
I think I've seen some people mention that `apt update` vs `apt-get update` might not be as problematic, but I haven't personally tested that myself.

jneilliii, 2021-09-10:
I noticed you had done this before [here](https://github.com/guysoft/CustomPiOS/commit/08a151906e32d4980155ce8cd42af39c424cbf3a) but that is no longer active, assuming that pi-gen updated their base image?

guysoft, 2021-09-12:
@jneilliii That was a workaround when Rapi foundation started upgrading packages to buster, which was done i a way where there were mixed packages, its written in the comment. 

Unlike the situation here, where that code is inside an if statement, only supposed to run in Ubuntu images. Not in Raspberrypi OS.

They might have accidentally pushed something from bullseye which they might also be working on, but I don't know that for sure.

jneilliii, 2021-09-12:
I think you missed the reply where I mentioned the patch to CustoPiZer was updated again outside of that if statement.

guysoft, 2021-09-12:
@jneilliii I did miss it, that corresponds to this line in CustomPiOS:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/disable-services/start_chroot_script#L12
At least I think that is the one, its the only place that looks like it.

Now that I know what line is cuasing it I could reproduce:

```
+ trap cleanup SIGINT SIGTERM
+ apt-get update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.7 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [393 kB]
Reading package lists...
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
```

Here is a talk on the rpi forms:
https://www.raspberrypi.org/forums/viewtopic.php?t=318302

Also opened an issue here:
https://github.com/RPi-Distro/pi-gen/issues/546