guysoft/CustomPiOS #145: Clarification of License Terms wrt. Artifacts and Custom Distributions

memchk, 2021-09-13:
For reference, I am considering producing a small-scale commercial product where the details would be closed source.

1. The CustomPIOS code itself is obviously under the GPLv3 terms as anyone would expect.

2. For my specific distribution modules / tree, is the intent for the GPL to be copyleft here? I.E if I make a "memchkOS" distro using the CustomPIOS framework would you that tree to be distributed with the GPL or compatible license?

3. As above, what is the license status of the built image, including any proprietary source code that may be included?

I guess this question comes down to if licensing terms like the GCC are expected where the compiler itself is under GPL, your code and the produced artifacts are unencumbered, or if the intent is to be completely copyleft?

guysoft, 2021-09-13:
1. Yes its is :)
2. the modules in your distro and your distro code is yours to license. You can supply the CustomPiOS code with it but it has to be under GPLv3. 
2. The license of the built image depends on what you are shipping in the image. Same as if you would have opened the image, edited it manually, and closed it up.

Its indeed similar to GCC, though note that many images have lots of requirements that might not be relevant to CustomPiOS.

If you have any more questions, or there is something in the current license that is limiting you please feel free to share.

memchk, 2021-09-13:
I think that clarifies it for me, I will close the issue now. Thanks for making CustomPIOS, it has turned out to be extremely useful in my prototyping!

closed 2021-09-13 by memchk