guysoft/CustomPiOS #150: No rule to make target install

cooked, 2021-12-31:
Hi,
I'm writing a module to build (and install) opencv from source. I'm not able to complete the installation step successfully, not with "make install" nor with "checkinstall" which returns the message below.
I believe I'm missing something fundamental here, and I'm looking forward to your input.

thanks
Stef

========================= Installation results ===========================
make: *** No rule to make target 'install'.  Stop.

****  Installation failed. Aborting package creation.

Cleaning up...OK

Bye.

++++ echo_red 'build failed, unmounting image...'
++++ echo -e -n '\e[91m'
++++ echo build failed, unmounting image...
build failed, unmounting image...
++++ echo -e -n '\e[0m'

guysoft, 2022-01-03:
1. I would suggest compiling OpenCV separately since it can take a few hours.
2. If you are not modifying OpenCV and just using it to write stuff you can install it using a package manager:
https://singleboardbytes.com/647/install-opencv-raspberry-pi-4.htm
3. If you still want to build OpenCV you need to install the dependencies and run Cmake which generates the makefiles, docs are here: https://docs.opencv.org/4.x/d7/d9f/tutorial_linux_install.html

cooked, 2022-01-03:
Hi Guy,
thanks for the feedback but even with that I'm not sure how to move on:
1) not sure what you mean exactly here. I've already compile opencv in other places and other projects, but the ultimate goal here is indeed to use a script that I already dialed in and I know it's working (when not in the custompios framework)... the script should be "set and forget" to some extent and allow github actions to build me an image overnight (when I launch it before going to bed let's say)... I already know first hand that the build takes ages
2) I've investigated that option and starting with Bullseye the OpenCV from the repo is a good start...however looking at the build log of the opencv packages available in the repo both the raspiOS distribution build openCV with the same configuration of the Debian one which is arm-generic and not optimized for the hardware of the various Pi boards, which is my target... so again, I'd rather compile opencv from source with my own config recipe
3) this is my choice. let me clarify... I already CAN build openCV, I have the dependencies and all... I CAN'T install it though, that's where I'm stuck

so the same question still applies:   how do I get "make install" to succeed? I might be missing something related to how custompios operates behind the scenes here...

cheers

guysoft, 2022-01-03:
1. I mean that recompiling opencv should not be part of the distro build step. Because it would happen all the time
2. Ok, if you know how to build it optimized feel free to share it, we could have a module set up
3.I understand now. I will need to see how you build it that it fails, the output you gave does not list the folder your are running in, what commands ran to build and what are the configure parameters. I need more information to understand what you are doing. Provide full reproduction steps.