guysoft/CustomPiOS #152: preserving permissions for '/boot/.': Operation not permitted

Joe2824, 2022-01-19:
I try to build an Image but the process ended with build failed. Because auf no permission to boot?
`cp: preserving permissions for '/boot/.': Operation not permitted

I attached the build log.
[build.log](https://github.com/guysoft/CustomPiOS/files/7901141/build.log)

guysoft, 2022-01-19:
It works fine on my other distros, here is example output on OctoPi.
I also used the image you are using ``2021-12-02-raspios-buster-armhf-lite.zip``

1. What build method are you use? eg are you using the docker image or github actions?
2. Can you share the code of your distro?

Without a basic reproduction code I can't figure out what is the issue.

```
+ cp /CustomPiOS/modules/network/start_chroot_script chroot_script
+ chmod 755 chroot_script
+ cp /CustomPiOS/common.sh common.sh
+ chmod 755 common.sh
++ arch
+ '[' x86_64 '!=' armv7l ']'
++ arch
+ '[' x86_64 '!=' aarch64 ']'
++ arch
+ '[' x86_64 '!=' arm64 ']'
+ '[' armv7l == armv7l ']'
+ echo 'Building on non-ARM device a armv7l system, using qemu-arm-static'
Building on non-ARM device a armv7l system, using qemu-arm-static
+ chroot . usr/bin/qemu-arm-static /bin/bash /chroot_script
+ set -e
+ export LC_ALL=C
+ LC_ALL=C
+ source /common.sh
+ install_cleanup_trap
+ set -e
+ trap cleanup SIGINT SIGTERM
+ unpack /filesystem/boot /boot
+ from=/filesystem/boot
+ to=/boot
+ owner=
+ '[' 2 -gt 2 ']'
+ mkdir -p /tmp/unpack/
+ cp -v -r --preserve=mode,timestamps /filesystem/boot/. /tmp/unpack/
'/filesystem/boot/./custompios-wpa-supplicant.txt' -> '/tmp/unpack/./custompios-wpa-supplicant.txt'
+ '[' -n '' ']'
+ cp -v -r --preserve=mode,ownership,timestamps /tmp/unpack/. /boot
'/tmp/unpack/./custompios-wpa-supplicant.txt' -> '/boot/./custompios-wpa-supplicant.txt'
+ rm -r /tmp/unpack
+ DIST_NETWORK_FILE=/boot/octopi-wpa-supplicant.txt
```

Joe2824, 2022-01-20:
I tried to build the image on a raspberry pi but I think the copy over wifi from my pc set the owner of the files wrong. I build now in Docker without any trouble.
Thanks for your fast reply 👍

closed 2022-01-20 by Joe2824