guysoft/CustomPiOS #154: Remove flavor config

guysoft, 2022-01-31:
Hey all,
To my knowloadge there is no one using the flavor build system, since you can just use the variant system.

Due to flag changes because of #151 I am alerting that I am planing to drop support for it.

If anyone has objections, is using it or has any other reason please object/support here.

guysoft, 2022-02-02:
@mrbeam I can see this in your repos, are you still using it?

guysoft, 2022-02-04:
Related https://github.com/mrbeam/BeamOS/issues/90

guysoft, 2022-03-20:
Got an answer from MR bean that they are using legacy CustomPiOS and don't mind. Also it seems like no one else is using it. Dropping support.

closed 2022-03-20 by guysoft

kraeml, 2022-06-09:
May you have to remove the paragraph in the wiki.

https://github.com/guysoft/CustomPiOS/wiki/Building#variant-flavors

guysoft, 2022-07-03:
Thanks, done