guysoft/CustomPiOS #156: How do you use the release script?

David00, 2022-02-16:
I have successfully implemented a pretty lengthy Custom Pi OS build process for an upcoming user interface launch, but I'm struggling to figure out the last step of building a custom image: the release script.

In my [last issue](https://github.com/guysoft/CustomPiOS/issues/126#issuecomment-846288257) quite some time ago, I got stuck in the same place, though that was a different Custom Pi OS build.

I'm not really sure how to go about executing the release script.

If I execute it directly from CUSTOM_PI_OS_PATH/src/release, it gives the following error about the `workspace` dir, and complains that it can't find the zip with raspbian in the name:

```
david@raspberrypi:~/CustomPiOS/src $ ./release
./release: line 14: pushd: workspace: No such file or directory
--> Building VARIANT default, FLAVOR default
Import the variant config if we have one
loading base config at /home/david/CustomPiOS/src/modules/base/config
ls: cannot access '/home/david/CustomPiOS/src/images/*-raspbian*.zip': No such file or directory
```

The image is not name raspbian anymore - it's actually raspios: `2022-01-28-raspios-bullseye-arm64-lite.zip`

Due to the workspace directory error, I tried copying the release and argparse scripts into my <custom image>/src folder, and running it from there.

This appears to do something, but the first issue is the same complaint about not being able to find `*-raspbian*.zip`.

I experimented with some of the `BASE_` environment variables and re-ran `release` (`BASE_IMAGE_PATH`, `BASE_IMAGE_RASPBIAN`), but none of the ones I tried seemed to make a difference. I'm really shooting in the dark here and would be happy to provide a simple write up for your FAQ on how to use the release script if I can get this working.

Other issues I'm seeing with the build process, which is probably user error: after a successful `build_dist`, the output image file in my `src/workspace` directory is not using the name or version I specified in my config. It has the same name as the stock image.

[release_failure.txt](https://github.com/guysoft/CustomPiOS/files/8077234/release_failure.txt)

guysoft, 2022-02-16:
Hey, on vacation so I will have limited availability till the 21st

1. You need to pass the variant arg to release. Look how its done nightly build script.
2. You have custompios checked out and you ate running thr command with the working directory being the CutstomPiOS, it should be run from the distribution as the working directory.
3. What distribution are you building?
4. I warmly recommended you use the docker building method,  its not manditory, but its build command runs both the build and release command correctly.

David00, 2022-02-17:
Thanks so much for replying while on vacation! And thanks for the notes on running the release script.

I am building from the current official Raspberry Pi OS Lite (64-bit ARM image) based on Debian Bullseye.

I'll try the docker method. I recall having issues with it before, but now that I have made a couple successful builds with your project, I will probably have better luck.  Can I build an ARM image via Docker on an x64 host? Or do I need to run the Docker build method on an ARM host? I've been using a Pi 4 to build, but I'd actually prefer running an x64 Ubuntu server VM on my server hardware for this.

David00, 2022-02-19:
So I tried the Docker build method on my Pi 4 and ran into an issue with `losetup`:

> losetup: 2022-01-28-raspios-bullseye-armhf-lite.img: failed to set up loop device: No such file or directory

I've seen your [Building With Docker](https://github.com/guysoft/CustomPiOS/wiki/Building-with-Docker) Wiki entry, and it says make sure the loop module is setup if necessary. I don't know if it's necessary or not, but I did run `sudo modprobe loop`.  In fact, here is exactly what I did:

Checkout the project...
```
cd ~/CustomPiOS/src/
./make_custom_pi_os -g docker_test
... the 2022-01-28-raspios-bullseye-armhf-lite.zip image downloads and new folders are made ... 
cd docker_test/src/
sudo modprobe loop
docker run --rm -v `pwd`:/distro --device /dev/loop-control guysoft/custompios:devel build
```

The output from the docker run command is attached.  Looks like part of the bash script is still looking for `*-raspbian*.zip` but I don't think that's the issue. Just FYI.

I did not customize any of the distro files after running `make_custom_pi_os -g`.  Also, here's a `ls /dev/loop*` in case it's useful: 
```
$ ls /dev/loop*
/dev/loop0  /dev/loop1  /dev/loop2  /dev/loop3  /dev/loop4  /dev/loop5  /dev/loop6  /dev/loop7  /dev/loop-control
```

[Docker Build Failed Log.txt](https://github.com/guysoft/CustomPiOS/files/8101746/Docker.Build.Failed.Log.txt)

guysoft, 2022-02-21:
Hey,
The issue ``failed to set up loop device: No such file or directory`` is the most annoying one in the current build system, its random. It hapends because docker or the device file don't update in time.

Related issue:  #55

To workaround I suggest you create a docker-compose.yml file and start that up.
If you get the loop device error try to stop/start the docker-compose. Usually it fixes it after a second run.

guysoft, 2022-02-23:
Also worth mentioning: https://github.com/moby/moby/issues/27886

David00, 2022-02-24:
Thank you Guy.  Building with docker-compose worked just fine for me on my Pi 4 running Raspberry Pi OS (Bullseye). This also resolved the original question I had when opening this issue, which was how to have your project build the .zip.

+1 for using docker-compose to get rid of any host environment uncertainties.

closed 2022-02-24 by David00