guysoft/CustomPiOS #159: boot : wrong PARTUUID with module readonly

maitredede, 2022-03-10:
Hi,

After building an image using `devel` branch, I saw that my images are not booting due to PARTUUID mismatch between `cmdline.txt` and `blkid`. The `/etc/fstab` entry is equal to value of `blkid`.

I saw that `readonly` module overwrites `/boot/cmdline.txt`.

Here is my module configuration : `base(mymod,mymod2,disable-services(mymod3),readonly)

guysoft, 2022-03-21:
Hey,
It does, I haven't really tested that module.
Perhaps @s00500 who wrote it might be able to figure out what is going on?

s00500, 2022-03-21:
Hey all, I think the reason it overrides partuuid is just because I was lazy and did not do a better way of editing the cmdline.txt file.... Ideally some nice sed find and replace would be used to adjust the config in that file....

Can you post your original cmdline.txt ?