guysoft/CustomPiOS #162: armbian banana_pi_m1

steviehs, 2022-03-18:
I want to build images for the bananapi m1. After digging around and learning how to setup a variant, I am now struggling at a point, where CustomPiOS wants to set up wireless stuff: 

cat: /etc/wpa_supplicant/wpa_supplicant.conf: No such file or directory and yes, there is a wpa_supplicant folder in the armbian but not that file. 

So, before I assume wrong procedure (I'd like to provide a pr for that), should I:

a) make this cat more robust? -> check if the file is there before cat'ing?
b) add an additional distro "armbian" which checks for that action?
c) do something else?

guysoft, 2022-03-18:
Do you have an example image I can have a look at?
There is an armbian variant that might need updating.

steviehs, 2022-03-18:
https://redirect.armbian.com/bananapi/Bullseye_current

This one....

guysoft, 2022-03-20:
Ok, fixed.
The armbian variant was not well maintained since I have no armbian device to test with. And no one is testing and reporting for it.

You can now build with it using
```
sudo docker exec -it exampleos-build build armbian
```
Notes the armbian variant at the end.

Note that I haven't tested any modules with it and some might need adaptation.
You can test with 
```bash
if [ "${BASE_DISTRO}" == "armbian" ]; then
fi
```

closed 2022-03-20 by guysoft