guysoft/CustomPiOS #163: Add support for new Raspberry Pi OS .xz. userlessimage

guysoft, 2022-04-13:
Hey,
The new Rpi image has no pi user and uses a .xz compression format.

I could not get to fix it because I was on holiday in Egypt till yesterday. And today I am starting a new job. So it might take a small delay untill I can fix it. Probably should find time by the end if the week.

If anyone has any input please share it here.

pablosproject, 2022-04-13:
Reading through the updates [here](https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/), it looks like we can set up a default user (although not really recommended) by putting a file called `userconf` in `/boot` at the first launch.  We could do a module for that that does basically this:

```
passwd_enc=$(echo 'mypassword' | openssl passwd -6 -stdin)
echo "username:$password_enc" >> /boot/userconf
```

Otherwise, I think the user will be prompted with the wizard at first launch, but I need to test it though.

pablosproject, 2022-04-15:
@guysoft I started some work here https://github.com/guysoft/CustomPiOS/pull/164, hope to finish over the easter weekend if I have the time.

pablosproject, 2022-04-15:
So, regarding the new Pi User removal, I tried re-building with the latest image of my images. CustomPiOS works as usual, but on the first boot there it prompts for account creation:
[image]
[image]

This looks good to me, as it forces users to set a secure password and user for the machine.
Although I see that there are some distros made with CustomPiOS that may need a default user to run, like OctoPi (I suppose). The idea is to create a module that adds a default user using the `userconf` file method, which is the recommended by the RaspberryPi foundation.

guysoft, 2022-04-23:
I think that for now if the default user module is enabled, it should remove the creation of the new user.
I might vote on changing the behavior in the future, but for now I am not sure how to adapt to what Rpi have decided to to. I think it might be best to wait and see how people are using the usage-specific distros.

guysoft, 2022-04-23:
Also I think ill pull the xz image changes tomorrow so at least images would get buikt

guysoft, 2022-04-25:
Ok, puhsed 5032d8b1977c1bc8504411005e52a948582fa24e c50d74bcef65b220f9801a7d27f9e2eb1ff6e73c  e88ec0a88f6bffb6cd938d6683f44283aa8adbab fbed8fc97d21c716b62041d27838a05ec5ddeb5a 47fc3fd88e81ff211b71a764f80eb8f97da29934 . So xz should work, but still need to handle the username change dialog.

KwadFan, 2022-04-27:
Hello Guy, after a littlebit of messing around with custoPiZer, I faced the same in my tests.
Here is my solution: https://github.com/KwadFan/MainsailOS-ng/blob/a77e1c7cea9c21c1fcc055f83914b7f89a91ae72/workspace/scripts/03_piconfig.sh#L27

Regards

guysoft, 2022-05-01:
Thanks a log for the snippet @KwadFan , tested and it works.
I don't want to commit it in just yet becasue for some reason automatic login is enabled. That means that when you boot on tty1 the user drops to the default user shell that has by default sudo capabilities.
Once I fix that I will push this to devel.
If you know why auto-login happens it would help me fix that too.

guysoft, 2022-05-01:
Something with the user genration is creaing:
``/etc/systemd/system/getty@tty1.service.d/autologin.conf``
And I can't find out what it is.

KwadFan, 2022-05-02:
Hello @guysoft ,
Yes, I am researching that also, this is in my opinion a bad decision from raspberry. I am also fighting against it like you. The root cause for this is here: https://github.com/RPi-Distro/raspi-config/blob/2ef9a3a194b30a471e611803e9b46239193b74a5/raspi-config#L1392
Also they seem to replace the default getty.service with autologin.service with the raspi-config package. But, I am working on this. Trying to reverse that with a rc.local or systemd service during this week. Will let you know if there is progress

guysoft, 2022-05-02:
@ghollingworth Could you perhaps shed any light on why when we use ``/boot/userconf`` auto-login is enabled?

When we boot `` /etc/systemd/system/getty@tty1.service.d/autologin.conf`` does not exist, but after we ``/boot/userconf`` is used the autologin file is created and the the tty1 terminal logs in to the user which has sudo privileges.

I tried running grep for the whole rootfs for ``autologin.conf`` and could not find what is doing this.

Thanks!

KwadFan, 2022-05-02:
So, @guysoft ,
Here is, what I came up with: https://github.com/KwadFan/MainsailOS-ng/blob/f63bea7f4cbcdf9e17b89a8825143a5850b32b8d/workspace/scripts/98_disable_autologin.sh

Feel free to abuse it :)

Regards

guysoft, 2022-05-02:
@KwadFan Decent workaround, however, it does strengthen the issue that we don't know where this is coming from.
Would really value input from @ghollingworth or anyone in @raspberrypi or @RaspberryPiFoundation because this is potentially a security hole for anyone running raspberrypi OS-based distributions. Some vendors might not even know this is happening in their headless machines.

tl;dr for newcomers - autologin is loaded when the a default pi-user is created, its undocumented and happends when using the userconf system (now the default in Rpi OS).

KwadFan, 2022-05-02:
Thanks :)

But, yes, eleminating the root cause is the better way. Its more intended to be a temporary patch on that hole...
Would like to see, some comments from the main devs. Improving security through user renaming is wonderful, but its a shot in its own crownjuwels to have nopasswd login after.

guysoft, 2022-05-04:
Update: It happens here: https://github.com/RPi-Distro/userconf-pi/blob/master/cancel-rename#L28
Got a reply in the pi-gen distro.

KwadFan, 2022-05-04:
Ok, that makes no sense to me why they choose option B2 instead B1 if rename is canceled ( what I think for what reason this script is intended ).

Are there plans to revert/patch that?

guysoft, 2022-05-10:
@KwadFan Actually rpi also see it as an issue that needs to be fixed.
See here:
https://github.com/RPi-Distro/pi-gen/issues/614#issuecomment-1117572968
and here:
https://github.com/RPi-Distro/userconf-pi/issues/2#issuecomment-1116060499

However they their plan is to fix it next release and not right away, and this is a security issue in my option.

KwadFan, 2022-05-10:
@guysoft So, yes.... even RPI Devs aren't unfailable xD.

Yes, you are right this is a huuuge Security Risk. What also comes to my mind in the end, if you dont remove the tty1 as serial console, like I in mainsailOS do, you could do worse things over serial also...

What is our option? Waiting to let them fix it or a quick and dirty workaround?

guysoft, 2022-06-16:
rightly pointed out by @KwadFan , the patch should
1. Be applied at the end script of base module to allow upgrades
2. Should make sure the file is there before patching to allow working with buster.

Will find time to write that or PRs are welcome for this

KwadFan, 2022-06-18:
@guysoft So, if PR's are welcome, here we go:

Prove of concept see:
https://github.com/KwadFan/MainsailOS/actions/runs/2519831513 with buster image.
https://github.com/KwadFan/MainsailOS/actions/runs/2519947783 with bullseye 32bit image

Will remove those branches after merge :) So, if you visit in future, it works! ;)

guysoft, 2022-09-09:
New change was commited 8 days ago:
https://github.com/RPi-Distro/userconf-pi/commit/c2892a50021fdf3a0ac4d97e08f03330ee5db333

New image with that change and that brakes https://github.com/guysoft/CustomPiOS/commit/ff73605024f08725091ec736c3c6c4ebb5f219dc patch was released two days ago

hrueger, 2022-10-27:
Hi,
I'm using the userless .xz image and it has worked fine so far. However, I need to add my default user to the `dialout` group.
If I run `usermod -a -G dialout glue` in a `start_chroot_script` of the `permissions` module, I get `usermod: user 'glue' does not exist`.

My config file looks like this:
```bash
export DIST_NAME=MyDist
export DIST_VERSION=0.0.1
export MODULES="base(disable-overscan,update,kiosk,permissions)"
export BASE_USER=glue
export BASE_USER_PASSWORD="redacted"
export BASE_ADD_USER=yes
export BASE_SSH_ENABLE=yes
export BASE_OVERRIDE_HOSTNAME=myDistHost
export BASE_IMAGE_ENLARGEROOT=1000
export BASE_ARCH=arm64
```

Any idea?

guysoft, 2022-10-27:
This fully works now, closing

guysoft, 2022-10-27:
@hrueger Why do you need to add that?
I haven't tested, but I think the user pi gets renamed, so you need to add the pi user to the groups you want it to be in.
Source at:
https://github.com/RPi-Distro/userconf-pi/blob/master/userconf#L3

closed 2022-10-27 by guysoft

hrueger, 2022-10-27:
I've been struggeling with usb devices and found that adding the user to `dialout` helps. I'll try it with the pi user, thanks for the hint.

guysoft, 2022-10-27:
That would make sense because ``/dev/ttyACM0`` which some 3D printers use is a file name that was initially for dial-up modems. So assuming you are talking about OctoPi installs.

hrueger, 2022-10-28:
It's me again ;-)
How do I re-enable auto-login? Is there a config option for that? I couldn't find any.
I tried running `sudo raspi-config nonint do_boot_behaviour B2` in a `start_chroot_script` but it does not work.

guysoft, 2022-10-28:
Its in the base module config settting:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/config#L25

hrueger, 2022-10-28:
Thanks for the fast response. I don't completely understand. My config looks like this:
`
export BASE_ADD_USER=yes
export BASE_USER=glue
export BASE_USER_PASSWORD="redacted"
`
do you mean that I should set `BASE_ADD_USER` to `no`?
This did not work, I'm being prompted to create a user when I boot the pi for the first time.

guysoft, 2022-10-28:
Do you want to be prompted to create a use on the first boot? if not, then set ``export BASE_ADD_USER=yes``.
 you wrote earlier "How do I re-enable auto-login"? I think I misunderstood that as removing the auto user creation.
 
 To enable auto login you will need to stop this line from running: https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/start_chroot_script#L65
 You can read the comment above it to why its there: https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/start_chroot_script#L59
 
 ATM there is no config switch for that, it can be added. Actually saw that as a security vulnerability, and it was treated as a bug by Rpi here: https://github.com/RPi-Distro/userconf-pi/issues/2

hrueger, 2022-10-28:
Thanks for the hints. Yes, I wanted to not be promted and auto-login.
I got it working by adding `sed -i 's|do_boot_behaviour B1|do_boot_behaviour B2|g' /usr/bin/cancel-rename` to a start_chroot_script`.
This is just reverts the patch.

commit ff736050 (2022-06-15): Fixes #163