guysoft/CustomPiOS #165: Bullseye wifi doest connect

rodrigo2019, 2022-05-31:
Hello, I'm using [fluiddpi image](https://github.com/fluidd-core/FluiddPI) that is based on your image. I have two PI's, the first one uses the Buster version and it works fine. My problem is with the Bullseye version, it list all available wifi but doesn't connect.
My image didn't came with the iptables lib, I installed by myself but didn't solved.

Could you give me some tips?

Thanks you

guysoft, 2022-05-31:
Can you provide a build.log output?

Thanks,

rodrigo2019, 2022-05-31:
where is the path for the file?

guysoft, 2022-05-31:
So please use the issue tracker of FluiddPi that maintain that image, this issue tracker is for building CustomPiOS images, not flashing and using them:
https://github.com/fluidd-core/FluiddPI/issues

closed 2022-05-31 by guysoft