guysoft/CustomPiOS #166: Files are not showing in the image

pbackx, 2022-06-17:
I am trying to build a custom Ubuntu Server image for Raspberry Pi. I started from the basic project and added the following to the config file:

```
    export BASE_DISTRO=ubuntu
    export BASE_ARCH=arm64
    export BASE_ZIP_IMG=../image/ubuntu-22.04-preinstalled-server-arm64+raspi.img.xz
```

Now I'm trying to added files. I have added some files in the root folder of the module that was generated for my project. I can see them in the build.log file, but I don't see them in the resulting image. I do see some "removed" messages in the log, that I'm not sure I should worry about or if this is some kind of normal cleanup operation.

Is there anything obvious I am missing?

Here are some lines from the build.log involving one of the files I'm trying to add:

```
+ cp -vr --preserve=mode,timestamps /home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem .
'/home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem' -> './filesystem'
'/home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem/root/usr' -> './filesystem/root/usr'
'/home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem/root/usr/lib' -> './filesystem/root/usr/lib'
'/home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem/root/usr/lib/systemd' -> './filesystem/root/usr/lib/systemd'
'/home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem/root/usr/lib/systemd/system' -> './filesystem/root/usr/lib/systemd/system'
'/home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem/root/usr/lib/systemd/system/reversessh.service' -> './filesystem/root/usr/lib/systemd/system/reversessh.service'
...
+ cp -v -r --preserve=mode,timestamps /filesystem/root/. /tmp/unpack/
'/filesystem/root/./README' -> '/tmp/unpack/./README'
'/filesystem/root/./usr' -> '/tmp/unpack/./usr'
'/filesystem/root/./usr/lib' -> '/tmp/unpack/./usr/lib'
'/filesystem/root/./usr/lib/systemd' -> '/tmp/unpack/./usr/lib/systemd'
'/filesystem/root/./usr/lib/systemd/system' -> '/tmp/unpack/./usr/lib/systemd/system'
'/filesystem/root/./usr/lib/systemd/system/reversessh.service' -> '/tmp/unpack/./usr/lib/systemd/system/reversessh.service'
...
+ cp -v -r --preserve=mode,ownership,timestamps /tmp/unpack/. /
'/tmp/unpack/./README' -> '/./README'
'/tmp/unpack/./usr/lib/systemd/system/reversessh.service' -> '/./usr/lib/systemd/system/reversessh.service'
...
+ rm -rfv filesystem
removed 'filesystem/root/usr/lib/systemd/system/reversessh.service'
removed directory 'filesystem/root/usr/lib/systemd/system'
removed directory 'filesystem/root/usr/lib/systemd'
removed directory 'filesystem/root/usr/lib'
removed directory 'filesystem/root/usr'
...
+ cp -vr --preserve=mode,timestamps /home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem .
'/home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem/root/usr' -> './filesystem/root/usr'
'/home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem/root/usr/lib' -> './filesystem/root/usr/lib'
'/home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem/root/usr/lib/systemd' -> './filesystem/root/usr/lib/systemd'
'/home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem/root/usr/lib/systemd/system' -> './filesystem/root/usr/lib/systemd/system'
'/home/backxp/dev/CustomPiOS/SasoDistro/src/modules/sasodistro/filesystem/root/usr/lib/systemd/system/reversessh.service' -> './filesystem/root/usr/lib/systemd/system/reversessh.service'
...
+ rm -rfv filesystem
removed 'filesystem/root/usr/lib/systemd/system/reversessh.service'
removed directory 'filesystem/root/usr/lib/systemd/system'
removed directory 'filesystem/root/usr/lib/systemd'
removed directory 'filesystem/root/usr/lib'
removed directory 'filesystem/root/usr'
```

guysoft, 2022-06-17:
Can you provide the unpack command you are using?

You need to tell CustomPiOS where to unpack the files and what permissions they should have.

Docs:
https://github.com/guysoft/CustomPiOS/wiki/Common.sh-functions#unpack

For Example:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/start_chroot_script#L16

pbackx, 2022-06-20:
So, I don't know what I did wrong, but I started from scratch to make a repro case and this is working:

https://github.com/pbackx/CustomPiOS/tree/devel/TestDistro/src

I compares my original code that failed with this one and I can't figure out the difference. I guess my attention wasn't there on Friday afternoon.

closed 2022-06-20 by pbackx

guysoft, 2022-06-20:
Ok, just to confirm you got it working, because it might be worth adding to the wiki some detail on how to get this working for others to have an easier time.

Also - I am not sure TestDistro should be forked from CustomPiOS. because CustomPiOS is used to generate that folder tree.