guysoft/CustomPiOS #176: Build in Github Actions not successful with Ubuntu-20.04.05

JuJup, 2022-09-09:
Hey there!
I started to tinker around with CustomPiOS and Github Actions and couldn't make it work.
After checking other repositories using CustomPiOS I could see, that the builds fail since September 7 (e.g. OctoPi).
The only difference I can see is the change of the ubuntu version in the runner. With 20.04.04 it seemed to work, now the runner is using 20.04.05. Unfortunately, I can't tell the runner which exact version it should use.

Furthermore, the error happens somewhere in the _/patches/fix-163-autologin-console.patch_. I could see projects, where this patch is not called and their build is running successfully.
Unfortunately, I just started digging into this project, so I have no ideas in mind how to fix this.
Maybe somebody can help :)

A part of the log before the error occurs:
`
Setup default user and password ...
+ echo -e -n '\e[0m'
++ get_os_version
++ local os_version
++ grep -c buster /etc/os-release
+ '[' 0 == 0 ']'
+ create_userconf
+ local pw_encrypt
+ '[' -n pi ']'
++ echo raspberry
++ openssl passwd -6 -stdin
+ pw_encrypt='$6$rmdZ8JK34QLLzWu1$vcar46INDuAj/0J2q./iQvCdbJaWxxMcBNc/kg72P3ICINRxN16HeUV9nAkqgHptxJvOPRac.m/fGNtgw696X0'
+ echo 'pi:$6$rmdZ8JK34QLLzWu1$vcar46INDuAj/0J2q./iQvCdbJaWxxMcBNc/kg72P3ICINRxN16HeUV9nAkqgHptxJvOPRac.m/fGNtgw696X0'
+ apt install --yes --only-upgrade userconf-pi

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
userconf-pi is already the newest version (0.5).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ unpack /filesystem/patches /patches root
+ from=/filesystem/patches
+ to=/patches
+ owner=
+ '[' 3 -gt 2 ']'
+ owner=root
+ mkdir -p /tmp/unpack/
+ cp -v -r --preserve=mode,timestamps /filesystem/patches/. /tmp/unpack/
'/filesystem/patches/./fix-163-autologin-console.patch' -> '/tmp/unpack/./fix-163-autologin-console.patch'
+ '[' -n root ']'
+ chown -hR root:root /tmp/unpack/
+ cp -v -r --preserve=mode,ownership,timestamps /tmp/unpack/. /patches
'/tmp/unpack/.' -> '/patches'
'/tmp/unpack/./fix-163-autologin-console.patch' -> '/patches/fix-163-autologin-console.patch'
+ rm -r /tmp/unpack
+ pushd /usr/bin
/usr/bin /
+ patch -p1
patching file cancel-rename
Hunk #1 FAILED at 25.
1 out of 1 hunk FAILED -- saving rejects to file cancel-rename.rej
++++ echo_red 'build failed, unmounting image...'
++++ echo -e -n '\e[91m'
++++ echo build failed, unmounting image...
`

Regards

guysoft, 2022-09-09:
Hey, related:
https://github.com/guysoft/FullPageOS/issues/440
The new Rpi image breaks the current build system because they changed the userless repair script.

See: #163

KwadFan, 2022-09-09:
Hey @guysoft ,
I allowed my self to do some of your fix work ;)

Happy Weekend and regards.

PR #177 (2022-09-09): fix: fixes build error in base module
This fixes #176

Location, linewise has changed in cancel-rename script.

Uses now a generic method to patch