guysoft/CustomPiOS #189: Error while building 64-bit image

tampe125, 2023-01-16:
Hello,

I'm currently trying to build an image from standard Raspberry Pi 64-bit image (2022-09-22-raspios-bullseye-arm64). However it fails with the following error:

```
++ uname -m
+ '[' x86_64 '!=' armv7l ']'
++ uname -m
+ '[' x86_64 '!=' aarch64 ']'
+ '[' armv7l == armv7l ']'
+ grep -q gentoo /etc/os-release
++ which qemu-arm-static
+ cp /usr/bin/qemu-arm-static usr/bin/qemu-arm-static
+ cp /home/tampe125/git/CustomPiOS/src/modules/base/start_chroot_script chroot_script
+ chmod 755 chroot_script
+ cp /home/tampe125/git/CustomPiOS/src/common.sh common.sh
+ chmod 755 common.sh
++ uname -m
+ '[' x86_64 '!=' armv7l ']'
++ uname -m
+ '[' x86_64 '!=' aarch64 ']'
++ uname -m
+ '[' x86_64 '!=' arm64 ']'
+ '[' armv7l == armv7l ']'
+ echo 'Building on non-ARM device a armv7l system, using qemu-arm-static'
Building on non-ARM device a armv7l system, using qemu-arm-static
+ grep -q gentoo /etc/os-release
+ chroot . usr/bin/qemu-arm-static /bin/bash /chroot_script
qemu-arm-static: /bin/bash: Invalid ELF image for this architecture
```
Any clues on that? Am I missing something or 64-bit images are not supported?
I'm currently using Ubuntu 22.04

guysoft, 2023-01-16:
Hey, what is the output of ``uname -m`` on your pi? and what is the output of ``arch``?
Because it looks like the pi is saying its using a 32bit system.

It might also be related to a commit that went in yesterday that changed the detection method:
https://github.com/guysoft/CustomPiOS/commit/765784422198fd1294cdd10b591666d5407d337b

cc @bwalex who made that commit if its relevant

tampe125, 2023-01-16:
Mhm wait, I'm building the image from my workstation, not from my pi, so I'm doing a cross compilation here.  
Do you still need the output from those two commands you posted?

guysoft, 2023-01-16:
A full log might help.

The error at the bottom is that its using qemu-arm-static which is not compatible with the /bin/bash in the image.
If you are building on ubuntu perhaps using the docker build method would be a good way to ensure that there is no environment irregularity.

hrueger, 2023-01-17:
@tampe125 did you add `export BASE_ARCH=arm64` to the `config` file? that solves the issue for me.

guysoft, 2023-01-18:
Thanks @hrueger forgot to ask that too.
See: 
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/config#L100

guysoft, 2023-04-25:
hmm, it looks like the arch is set to 64bit.
I am pushing a fix, make sure that you set in config:
```bash
export BASE_ARCH=armv7l
```

Could you test that it works for you?

guysoft, 2023-08-21:
Any news on this? Closing if no reply.

dirkhh, 2023-08-21:
My issue is solved, but I'm not the original poster...

guysoft, 2023-08-21:
Will wait to see if @tampe125 has an answer, and if not will close it

guysoft, 2023-08-21:
Ok, since we think its fixed closing for now. If you change your mind feel free to re-open the issue.

closed 2023-08-21 by guysoft