guysoft/CustomPiOS #190: Change repo branch on nightly build

calonmerc, 2023-02-16:
I'd like to leave the primary `release.yml` to pull a repo master, but the nightly `build.yml` to pull the same repo but on development.

Currently, images are building fine, but only pulling the master branch. I've been searching and testing on and off for a week. Wondering if there's a simpler way, maybe there's a way to know what *.yml is starting the build process? 

I've looked through all existing modules in all listed builds based on CustomPiOS, but none are doing what I'm looking for. 

[Here's the repo, including the current GH action files](https://github.com/calonmerc/PiFireOS)

guysoft, 2023-02-16:
I am not sure repo branch you are referring to. The CustomPiOS docker image branch? or your own branch?

You can decide which yaml would run on which brach on your repo:
https://stackoverflow.com/a/72601918/311268

calonmerc, 2023-02-16:
I'm referencing the git clone in the start_chroot_script within my repo.

calonmerc, 2023-02-16:
https://github.com/calonmerc/PiFireOS/blob/af0412138a2d0fc842584681ce63a19d965f8290/src/modules/pifire/start_chroot_script#L46

guysoft, 2023-02-16:
You can create a ``config.local`` file in the distro folder and set environment variables from there that are specific to the build. Then you can set the variable you created there ``PIFIRE_REPO_BRANCH``

```bash
echo PIFIRE_REPO_BRANCH=devel >> /PiFireOS/src/config.local
```

Also you are welcome to add PiFireOS to the CustomPiOS readme list :)
https://github.com/guysoft/CustomPiOS#list-of-distributions-using-custompios

calonmerc, 2023-02-16:
Looks like that's what I was missing! Will report back after this test build runs!

I planned to request being added after I got the nightlies running on dev! Is there a proper process, or just follow the requests of others as a new issue?

calonmerc, 2023-02-18:
I never reported back... that was definitely the fix I was looking for!

closed 2023-02-18 by calonmerc