guysoft/CustomPiOS #195: Fails to resize partition after April 24 release

sunefred, 2023-04-25:
This was discovered in my CI/CD build running on GitHub. The agent is running [ubuntu-22.04](https://github.com/actions/runner-images/blob/main/images/linux/Ubuntu2204-Readme.md).

Here is a snippet of a working "losetup" when using the older version

```
+++ detach_all_loopback 2022-09-22-raspios-bullseye-armhf-lite.img
++++ losetup
++++ awk '{ print $1 }'
++++ grep 2022-09-22-raspios-bullseye-armhf-lite.img
+++ test_for_image 2022-09-22-raspios-bullseye-armhf-lite.img
+++ '[' '!' -f 2022-09-22-raspios-bullseye-armhf-lite.img ']'
++++ losetup -f --show -o 272629760 2022-09-22-raspios-bullseye-armhf-lite.img
+++ LODEV=/dev/loop3
+++ trap 'losetup -d $LODEV' EXIT
+++ e2fsck -fy /dev/loop3
e2fsck 1.46.2 (28-Feb-2021)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
rootfs: 47607/97920 files (0.1% non-contiguous), 296222/391168 blocks
+++ resize2fs -p /dev/loop3
resize2fs 1.46.2 (28-Feb-2021)
Resizing the filesystem on /dev/loop3 to 5511168 (4k) blocks.
The filesystem on /dev/loop3 is now 5511168 (4k) blocks long.
```

And this is the same code, but failing, from the April 24 release

```
+++ detach_all_loopback 2022-09-22-raspios-bullseye-armhf-lite.img
++++ grep 2022-09-22-raspios-bullseye-armhf-lite.img
++++ awk '{ print $1 }'
++++ losetup
+++ test_for_image 2022-09-22-raspios-bullseye-armhf-lite.img
+++ '[' '!' -f 2022-09-22-raspios-bullseye-armhf-lite.img ']'
++++ losetup -f --show -o 272629760 2022-09-22-raspios-bullseye-armhf-lite.img
+++ LODEV=/dev/loop3
+++ trap 'losetup -d $LODEV' EXIT
+++ file -Ls /dev/loop3
+++ grep -qi ext
/CustomPiOS/common.sh: line 300: file: command not found
+++ file -Ls /dev/loop3
/CustomPiOS/common.sh: line 303: file: command not found
+++ grep -qi btrfs
```

For now I am staying with the older release by specifying it by digest as follows

```
docker run --rm --privileged --volume "distro:/distro" --device /dev/loop-control guysoft/custompios@sha256:fa636e7b2696fb7c89580a147838f00c2644ffc012211408ba70947cb2745b7d build
```

sunefred, 2023-04-25:
Adding @dirkhh, the #194 contributor.

guysoft, 2023-04-25:
Great, thanks for reporting

guysoft, 2023-06-11:
BTRFS works now but does not resize correctly

closed 2023-06-11 by guysoft