guysoft/CustomPiOS #197: RPM based distro support

MrMEEE, 2023-05-23:
Hi

I'm about 90% of the way, implementing support for Rocky and CentOS (maybe RHEL in time) into CustomPiOS..

Is this something that you guys would be interested in me upstreaming? or would you prefer that I maintain a separate fork?

guysoft, 2023-05-23:
Gladly! Thanks so much for sharing and offering! 
Though I would like to see how you go about with it so it does not clash with the other existing workflows. Also it might require some documentation.

MrMEEE, 2023-05-24:
Of course.. 

For now I have only tested with Rocky (there are no official images for RHEL and CentOS, but I plan to fix that).. 

I will see if I can get a Pull request done in the next couple of weeks

guysoft, 2023-08-20:
Hey, any progress on that?