guysoft/CustomPiOS #199: Bug /dev/loop0 Partition don't follow the config to resize

Neko-vecter, 2023-07-03:
After commit `71787f060221069848abfcbb307f616680f7dbb1`. Cannot create a larger partition for img using ubuntu server 23.04.
Starting at `06d9bdbb7077a792c7a5057e9a44cc384dfc87c3` it is not possible to use the configuration to create a larger disk partition.

I think it might be related to the modification of `common.sh`. Based on the commit logs, support for Btfs was added in `06d9bdbb7077a792c7a5057e9a44cc384dfc87c3` version.

System info.
[image]

guysoft, 2023-08-20:
Hey, what image are you building? Could you provide a build log so I can determine if there some edge case where it detects what filesystem its resizing? That code does stuff to resize BTRFS systems, but its likely you are using EXT4

Neko-vecter, 2023-08-20:
> Hey, what image are you building? Could you provide a build log so I can determine if there some edge case where it detects what filesystem its resizing? That code does stuff to resize BTRFS systems, but its likely you are using EXT4

---

The image I am trying to building is mainsailos. 
The img I am using is Raspbian 64bit ver. (It seems to be a EXT4 img)

---

I think it's the if statement in `common.sh` that's causing the problem.

The code below is the screenshot of `common.sh`. The line `300` if statement it was skipped for unknown reasons. (not sure if I miss some dependencies to cause this issue). 

[image]

The code below is missing in this if statement.
``` shell
e2fsck -fy $LODEV
resize2fs -p $LODEV
```

Then I swap the function `function enlarge_ext()` to previous version that work. The partition resize normality.

It seems this issue can be solved by adding a fallback. Like add `else` at the end? and run the code that EXT4 need?

---

Also, what dependence I need? The dependence I install "aria2, coreutils, jq. p7zip-full, qemu-user-static, zip". Or? Do this is somethings only in GitHub Action?

-Neko.vecter

Neko-vecter, 2023-08-21:
> Ok, pushed an echo that should print the filesystem line result if it can't determine, could you build and paste that result? That way we can figure why its failing.

??? what?
[image]

Neko-vecter, 2023-08-28:
Since problem solved... I think we can close this issue?

Conclusion:
This issue cause by missing dependence that `common.sh` need. The dependence `file` is missing by system installation way (Ubuntu minimal). 

-Neko.vecter

closed 2023-08-28 by Neko-vecter