guysoft/CustomPiOS #201: qemu_boot requires a compressed image in /tmp

umlaeute, 2023-08-31:
i'm just starting with CustomPiOS, and i'm a bit stumped on how to use `qemu_boot`.

1. no help or whatsoever
   it would be really nice if starting `qemu_boot.sh` without any arguments (or with `-h`; or with a non-existing file) to output some usage information
2. it apparently requires the image to be passed either as ZIP-compressed archive or an XZ-compressed image - why is it not possible to use a raw (uncompressed) image, such as created by `build_dist`
3. when using a ZIP-compressed archive the image has to be the first entry in the archive (I don't know whether this is also a requirement by rpi-imager, but it would be nice to have this documented)
4. when using an XZ-compressed image, it *must* live in `/tmp` otherwise the uncompressed image won't be found

umlaeute, 2023-08-31:
also, from checking the code it seems that the image-name (unless it is a zip-file) *must* not contain the sequence `xz` preceded by an arbitrary character: https://github.com/guysoft/CustomPiOS/blob/f6fea14adfe9febc288bd7c6db8466755a81825f/src/qemu_boot.sh#L15

e.g. 

```sh
$ echo mixzap.img.xz | sed "s/.xz//"
map.img.xz
$
```

the correct sed-expression would be `"s/\.xz$//"`, but  i think a much better way would be to use POSIX parameter expansion:
```sh
IMG_NAME=$(basename "${ZIP_IMG%.xz}")
```

guysoft, 2023-08-31:
Hey, you are quite right. I initially wrote qemu_boot as a quick hack and it really has the place to become something more useful and well documented.
I am not sure how much time I have to do that now. Perhaps adding an if args<0 and to print help would be a good start. I would really welcome a PR if you have time for that.

umlaeute, 2023-08-31:
i might :-)

sidenote: shouldn't `qemu_boot.sh` and `qemu_boot64.sh` be virtually identical (obviously apart from the CPU/kernel/dtb/...)?

guysoft, 2023-09-02:
> i might :-)
> 
> sidenote: shouldn't `qemu_boot.sh` and `qemu_boot64.sh` be virtually identical (obviously apart from the CPU/kernel/dtb/...)?

Yes, though I update which kernel is used since they changed over time. They could be joined in to one script

umlaeute, 2023-09-07:
i found https://github.com/bablokb/pi-qemu-helper.git which seems to be a better alternative to your scripts (as it also supports booting with a GUI)

guysoft, 2023-09-07:
Cool - if you have some example of what commands to run on a .zip file to get a running emulation we could add that to the wiki.
Also I am interested too :)