guysoft/CustomPiOS #202: use POSIX parameter expansion for assigning default values

umlaeute, 2023-08-31:
the modules are full of default-assignments like so:
https://github.com/guysoft/CustomPiOS/blob/f6fea14adfe9febc288bd7c6db8466755a81825f/src/modules/base/config#L13

this is obviously not a bug, but i think the readability could be improved by using POSIX parameter expansion.

the following is equivalent to the test-clause above:

```sh
: ${BASE_RELEASE_COMPRESS:=yes}
```

from the bash manpage:
>  `${parameter:=word}`   **Assign  Default Values**.  If *parameter* is unset or null, the expansion of *word* is assigned to *parameter*.  The value of *parameter* is then substituted.  Positional parameters and special parameters may not be assigned to  in this way.

guysoft, 2023-08-31:
So it would be in the config file:
```bash
BASE_IMAGE_RASPBIAN=${BASE_RELEASE_COMPRESS:=yes}
```
?

I want to assign default values there because its also where people go to look what they can set.

umlaeute, 2023-08-31:
i'm not sure what you mean.
your example would
1. set `BASE_RELEASE_COMPRESS` to `yes` unless it is already set to a non-empty value
2. set `BASE_IMAGE_RASPBIAN` to the value of `BASE_RELEASE_COMPRESS` (from item1)

i guess that mixing of `BASE_IMAGE_RASPBIAN` and  `BASE_RELEASE_COMPRESS`  was just a typo.

my line:
```bash
: ${BASE_RELEASE_COMPRESS:=yes}
```
does:
1. set `BASE_RELEASE_COMPRESS` to `yes` unless it is already set to a non-empty value
2. nothing else (as `:` is a nop-command)

you can of course use variables on the right-hand term as well:

```sh
: ${BASE_IMAGE_PATH:=${DIST_PATH}/image}
```

but if you feel that this is just busy-work, feel free to ignore (and possibly close) this ticket