guysoft/CustomPiOS #205: allow out-of-tree builds

umlaeute, 2023-09-14:
i find it slightly annoying that running `build_dist` will put the `workspace/` directory into the same directory where the `build_dist` script lives.

it would be nice if i could clone my `CustomPiOS` repo into my (small, fast) home, and do:

```sh
git clone github.com:me/MyCustomPiOS
mkdir /media/big_disk/MyCustomPiOS_build
cd /media/big_disk/MyCustomPiOS_build
~-/MyCustomPiOS/mydist/src/build_dist
ls -lha workspace/
```
---

this might be a problem with how i customize CustomPiOS though.
currently (after having checked out some of the forks) I leave the `src/` folder as is, and put all my (git-tracked) modules/config/... into `mydist/src/`.

alternatively, I could of course put all my stuff into `src/`, and turn the build-process into something like
```sh
src/make_custom_pi_os -g -v raspios_full_armhf mydist
mydist/src/build_dist
```

but that would mean that i need to modify `src/dist_generators/dist_example/` to use sane defaults for `mydist`, and i somewhat dislike this approach.

anyhow, it's probably just a matter of documentation :-)

guysoft, 2023-09-14:
It can be changed here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/config#L48

Be warned I haven't tested that in a while

closed 2023-09-14 by guysoft

umlaeute, 2023-09-14:
thx, actually i think i tried that bt it seems it didn't work.
i'll re-open once i think it is important enough for me :-)