guysoft/CustomPiOS #209: Build within container abruptly stops on current `devel` image

David00, 2023-10-09:
Hi @guysoft,

I am in the process of moving my build environment to a new server, and this process has me chasing a strange issue that seems to be related to the current docker image.

Here are the steps to reproduce (make sure that you don't have a cached `guysoft/custompios:devel` docker image already!)

```bash

cd /tmp
git clone https://github.com/guysoft/CustomPiOS.git
cd CustomPiOS
./src/make_custom_pi_os -g test_distro

cd test_distro/src/

echo """version: '3.6'

services:
  custompios:
    image: guysoft/custompios:devel
    container_name: test_distro_builder
    tty: true
    restart: always
    privileged: true
    volumes:
      - ./:/distro
    devices:
      - /dev/loop-control""" >> docker-compose.yml

docker compose -f docker-compose.yml up -d

docker exec -it test_distro_builder build
```

As you can see, this is a 100% stock pull and implementation as described in the docs.  The output above ends with the following lines:

```
+++++ git -C /CustomPiOS rev-parse HEAD
fatal: not a git repository (or any of the parent directories): .git
++++ BASE_COMMIT=
+ exit 1
```

I've tried to debug this through the various bash scripts but couldn't really figure it out.  There are no error messages raised, and the git message that says `fatal: not a git repository ... ` shows up in both the working build and the non working build. 

Here's the full [build.log](https://github.com/guysoft/CustomPiOS/files/12851031/build.log).

This build is happening with the current `devel` image on amd64.  Here's my `docker image inspect` output:
[docker image inspect.txt](https://github.com/guysoft/CustomPiOS/files/12851065/docker.image.inspect.txt)

I noticed that my image SHA doesn't match the SHA shown on docker hub so that was kind of concerning too.  The image where things are working fine was built on 9/9/2022, and has a `docker image inspect` that starts like:

```
[
    {
        "Id": "sha256:00fb8e40f27ca745780215286b06fd65e022d960c81e610bd59a741c7de64301",
        "RepoTags": [
            "guysoft/custompios:devel"
        ],
        "RepoDigests": [
            "guysoft/custompios@sha256:0c51743cc99289d9e1915735288672b3831be5f55c325ad7d70b96f2f1d1ee2a"
        ],
        "Parent": "",
        "Comment": "buildkit.dockerfile.v0",
        "Created": "2022-09-09T16:37:12.368800052Z",
```

Here is the build.log from the image described above, using the same exact setup steps I shared at the top of this issue (except I already have the old custompios:devel image.

[working_build.log](https://github.com/guysoft/CustomPiOS/files/12851146/working_build.log)

Thanks!

guysoft, 2023-10-10:
Hey, its caused by a regression form https://github.com/guysoft/CustomPiOS/pull/208
I am adding a workaround for now so it does build

closed 2023-10-10 by guysoft

David00, 2023-10-10:
Thank you for the super quick response and fix! I just tested the new image and it's good to go now.

guysoft, 2023-10-11:
FYI bookrowm was released yesterday and is likely to break

commit 55a92830 (2023-10-10): Fix regression caused by #208 , should fix #209