guysoft/CustomPiOS #211: permissions of the generated image

umlaeute, 2023-10-18:
the generated image file is *readable*, *writable* **and** *executable* by everybody:

```terminal
$ ls -lha workspace/*.img
-rwxrwxrwx 1 root     root      14G Oct 17 15:05 workspace/custompios.img
```

is this intended?

in general i heartily:
- making things writable by *others*
- making arbitrary data-files exectuable

the culprit can be found in  https://github.com/guysoft/CustomPiOS/blob/55a92830d166ba1f9c6702a3d9d8a20ed416f723/src/custompios#L207

i think this should be changed to

```sh
chmod 644 "${BASE_IMG_PATH}"
```

umlaeute, 2023-10-22:
- well `chmod 644` is still *readable* by user, group and others.
- do you need the image to be *writable* by others?
- i cannot imagine a workflow that requires the image to be *executable*

guysoft, 2023-10-22:
I agreed with you :)
You are welcome to send a PR

closed 2023-10-23 by guysoft