guysoft/CustomPiOS #212: Script ./src/make_custom_pi_os returns unexpected error

VovaStelmashchuk, 2023-10-20:
I start works with repo and just flow the tuturial from ReadMe file. Clone the repo and try to execute
`./src/make_custom_pi_os -g test` the script returns the following error:
```bash
Settings:
making dstro in test
variant: raspios_lite_armhf
./src/make_custom_pi_os: line 36: MULTIPLE[@]: unbound variable
```

I have no idea, about how to fix the problem. Sure i can just remove the part of script which produce the problem, but the way doesn't look right for me.
```bash
for a in "${MULTIPLE[@]}"; do
    echo "  $a"
done
```

My envirment
MacOs sonoma: Version 14.0
Processor: 2,4 GHz 8-Core Intel Core i9

guysoft, 2023-10-22:
Not sure what is going on. Will try and reoroduce.

umlaeute, 2023-10-23:
i cannot reproduce (on Debian/sid).

it might be interesting which `bash` you are using.

```sh
$ which bash
/usr/bin/bash

$ bash --version
GNU bash, version 5.2.15(1)-release (x86_64-pc-linux-gnu)
```

having said that: the `${MULTIPLE}` variable seems to be unused, except for the problematic line https://github.com/guysoft/CustomPiOS/blob/63da54b86ab566c558e9084568d326413f6585d8/src/make_custom_pi_os#L34

so it probably *is* safe to remove that part of the script

umlaeute, 2023-10-23:
i got hands of a macOS/BigSur machine, and it shows i can now reproduce the problem.

```sh
$ which bash
/bin/bash

$ bash --version
GNU bash, version 3.2.57(1)-release (arm64-apple-darwin21)
```

in the bash changelogs i find

```
[...]
This document details the changes between this version, bash-5.2-alpha, and
the previous version, bash-5.1-release.
[...]
bb. Array references using `@' and `*' that are the value of nameref variables
    (declare -n ref='v[@]' ; echo $ref) no longer cause the shell to exit if
    set -u is enabled and the array (v) is unset.
```

so maybe this is just requiring a too-new-for-macOS bash version (for something that doesn't get used anyhow...)

VovaStelmashchuk, 2023-10-23:
My current bash 3.2.57, I make a clean setup and attech all my logs here:
```bash
➜  hacklab git clone https://github.com/guysoft/CustomPiOS.git
Cloning into 'CustomPiOS'...
remote: Enumerating objects: 4837, done.
remote: Counting objects: 100% (810/810), done.
remote: Compressing objects: 100% (272/272), done.
remote: Total 4837 (delta 590), reused 594 (delta 524), pack-reused 4027
Receiving objects: 100% (4837/4837), 1000.27 KiB | 4.15 MiB/s, done.
Resolving deltas: 100% (2589/2589), done.
➜  hacklab cd CustomPiOS
➜  CustomPiOS git:(devel) which bash
/bin/bash
➜  CustomPiOS git:(devel) bash --version
GNU bash, version 3.2.57(1)-release (x86_64-apple-darwin23)
Copyright (C) 2007 Free Software Foundation, Inc.
➜  CustomPiOS git:(devel) ./src/make_custom_pi_os -g test
Settings:
making dstro in test
variant: raspios_lite_armhf
./src/make_custom_pi_os: line 36: MULTIPLE[@]: unbound variable
➜  CustomPiOS git:(devel)

VovaStelmashchuk, 2023-10-23:
@umlaeute you right, the error throws because i use the old version of bash, currently I update my bash and has the following output. Does the output correct ?
```
➜  CustomPiOS git:(devel) bash --version
GNU bash, version 5.2.15(1)-release (x86_64-apple-darwin23.0.0)
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
➜  CustomPiOS git:(devel) ./src/make_custom_pi_os -g test
Settings:
making dstro in test
variant: raspios_lite_armhf
sed: 1: "config": command c expects \ followed by text
➜  CustomPiOS git:(devel) ✗
```

VovaStelmashchuk, 2023-10-23:
Thx, the problem fix, after I install `gnu-sed`. 
Now I have problem, with execute `sudo ./test/src/build_dist`, I have the following error, in `build.log`
```bash
++++ realpath -s /Users/vovastelmashchuk/Desktop/hacklab/CustomPiOS/src/build
realpath: illegal option -- s
usage: realpath [-q] [path ...]
+++ dirname
usage: dirname string [...]
++ BUILD_SCRIPT_PATH=
You have new mail in /var/mail/vovastelmashchuk
```

It's look like a new issue, can we discuss the issue here or close current discussion and open new github issue ?
Also I think about pull request to project readme with MacOs specific instraction, some think like a pre-condition, min version of bash and gnu-sed

umlaeute, 2023-10-23:
i agree it's probably a new issue.

otoh, it seems that CustomPiOS is probably just not usable on macOS right now.

guysoft, 2023-10-23:
Indeed, you could use it via Docker. In general Docker tends to work better if you are on MacOS.

guysoft, 2023-10-31:
You could install bash 5 using homebrew.
```
brew install bash
```
But then you need to put bash in your path:
```
/opt/homebrew/bin/bash --version
```

VovaStelmashchuk, 2023-10-31:
Yep, I have bash 5 already on my machine. It fixes the issue. Thx.

closed 2023-10-31 by VovaStelmashchuk

guysoft, 2023-10-31:
Thanks! If you have the time to document or post here in the wiki how to gets stuff working on mac that could help people in the future (or yourself too):
https://github.com/guysoft/CustomPiOS/wiki/Building#building-from-os-x