guysoft/CustomPiOS #214: CustomPiOS v2 - support for remote modules, meta modules

guysoft, 2023-11-11:
Hey all,
So I am working on the idea of having CustomPiOS get better support for other boards, and modules that are stored on other repositories.
I still want to keep CustomPiOS "simple" so I am trying to limit the new logic. I would value your input if its too simple. Or not clear enough. Or if something critical is missing.

I have now a feature branch for meta modules.
The branch at work is here:
https://github.com/guysoft/CustomPiOS/tree/feature/meta-modules

This work is geared towards having what Mainsail did work on other distributions (namely OctoPi, but hopefully others too, perhaps FullPageOS).
Mainsail build for multiple boards using a "config" folder and a bunch of scripts to place the builds.
You can see how that looks here:
https://github.com/mainsail-crew/MainsailOS/tree/develop/config

### What are meta modules?
The idea is that you have a "meta" file in a modules. Which lets you add in sub modules based on if they are needed in a board. 
Its an executable script that can hold any logic you like. And example for one for base is here (and there is one for network:
https://github.com/guysoft/CustomPiOS/blob/feature/meta-modules/src/modules/base/meta
So for example:
```
base(octopi)
```
Could be turned by a meta file to
```
base(armbian(octopi))
```
### What are remote modules?
The idea is that you can have modules that are stored on other repositories, currently implemented only for git repositories, and have them downloaded and used at build time.
They are configured in this yaml file:
https://github.com/guysoft/CustomPiOS/blob/feature/meta-modules/src/modules_remote.yml
I could add a local yaml that a repo can add in their repository that would let you set extra modules per-repo.

### More info about what mainsail are doing
Partly this started because I am trying to organize what Mainsail are doing here:
https://github.com/mainsail-crew/MainsailOS/blob/develop/config/orangepi/default#L26
Namely how we handle base for armbian and orangepi and other boards

guysoft, 2024-01-02:
Just updating that there is initial support for this on the branch. I Want to build a multi-build command, that would let you build for multiple devices your OS. It also downloads the images and all together. Perhaps also have better CLI feedback.

Would value input for this.

geeklimit, 2024-01-15:
Sounds great, and I can test this today on a Raspberry Pi Zero W and a Libre Le Potato AML-S905X-CC. Unfortunately I'm not a Linux/software dev and I get a little lost on what to do.

For FullPageOS on the Libre board from "out of the box" to up and running on wifi, I'll try the following as a test?

-1. Install Raspbian, CustomPiOS, Ubuntu, or Debian (Armbian?) OS on the Le Potato
0. sudo apt-get install gawk util-linux realpath qemu-user-static p7zip-full git (xzcat? see step 11)
1. git clone https://github.com/guysoft/CustomPiOS.git
2. ./src/make_custom_pi_os -g armbian
3. cd armbian/src
4. sudo nano armbian/src/config
5. sudo ./armbian/src/build_dist fullpageos
6. df
7. (insert SD card)
8. df
9. (note new entry, as in '/dev/sdb1')
10. sudo umount /dev/sdb1
11. sudo xzcat ./armbian/src/build_dist/workspace/armbian.xz | sudo dd bs=4M of=/dev/sdb
13. sync
14. (power down, remove SD card reader, switch SD cards, power up and log in)
15. sudo iwlist scan | grep SSID
16. ifconfig
17. sudo nano /etc/network/interfaces
18. paste:

allow-hotplug (device ID from ifconfig, as in: wlp2s0)
iface wlp2s0 inet dhcp
wpa-ssid (mySSID)
wpa-psk (mySSIDpassword)

19.  sudo nano ./boot/fullpageos.txt
20. (enter web address to load in Chromium, save)
21. (Power down, connect headless to display, power up and test for 24+ hours, monitor for Chromium crashes or freezes)

guysoft, 2024-01-22:
@geeklimit
Sounds cool. And I whink tis worth a test, but there is no armbian image ofr librepotaton on yet. Once there is you can change this command to the right board+distro name:
``./src/make_custom_pi_os -g armbian``
And this:
``sudo xzcat ./armbian/src/build_dist/workspace-armbian-distro-name-path-to-be/armbian.xz | sudo dd bs=4M of=/dev/sdb``

geeklimit, 2024-01-22:
> @geeklimit Sounds cool. And I whink tis worth a test, but there is no armbian image ofr librepotaton on yet. Once there is you can change this command to the right board+distro name: [...]

@guysoft so this is no good, then? https://www.armbian.com/lepotato/

I could use Debian, I guess?

guysoft, 2024-01-22:
It should work with a little tweaking. The goal ia to have it and I have lepotato here donated for testing

geeklimit, 2024-01-22:
ok, I don't want to push it with the variables for a test, I'm happy to test with whatever - Debian, Ubuntu, etc.. I'll give it a try this afternoon.

meteyou, 2024-01-26:
@guysoft sry for the late answer. in case of le potato, i would recommend you to use the original image. Maybe @KwadFan can tell you the current state of the le potato image in our test workflow.

but i can add the image to our armbian build workflow, if you want to test it.

KwadFan, 2024-01-26:
@guysoft Hey, nice to ask me.

So, I am in closer contact with one of the developers of Libres interpretation of "Raspberry PiOS" (which it isnt really but based on some of the packages).

I was during my test confronted with a thing, how they handle their DTBs/Overlays. I had multiple back and forths with him and they decided to hear me. Therefore, there is now a posibilty to enable them in an easy way and I can built a sort of first boot service to enable "MainsailOS" typical setup.
But I got stuck due lack of time, so its in backlog and I will provide based on there Image a MainsailOS version.
That said, it should be easy to use for other projects then. I highly recommend using there own Images.
But it comes with the downside, they dont want to drop BTRFS as main `/` - FS. Even he tried to provide an api to resize the image if not "hot" mounted, I wasnt able to use. In the end, I see no benefit using armbian for this little boi ;)

Regards Kwad

guysoft, 2024-02-05:
Adding BTRFS for librepotato makes sense. I mean I would like to see that running.
I even asked about it over 10 years ago on stack exchange, it looked like a more stable option for SBCs at the time. But Rpi didn't support it:
https://raspberrypi.stackexchange.com/questions/8265/btrfs-root-filesystem-on-raspbian

@dsx724 I am not sure if you are the right address, but saw you active there and thought you might know who to ask. What image should we add to the image build index?

guysoft, 2024-02-06:
@KwadFan I see you had this effort at the time:
https://github.com/mainsail-crew/MainsailOS/pull/250/files

KwadFan, 2024-02-06:
> @KwadFan I see you had this effort at the time: https://github.com/mainsail-crew/MainsailOS/pull/250/files

Yes, and I didnt give up on that, had to move it to backlogs to figure out some things, dsx had changed quite a few things that you can use there overlay technic during first boot. Will pickup as soon I finished my current project. Hopefully only a few days :)

dsx724, 2024-02-09:
Sorry I was sick. I opened an issue years ago as well but RPi doesn't care: https://github.com/raspberrypi/linux/issues/1761 

@guysoft We have created this bugger: https://github.com/libre-computer-project/libretech-slipstream

guysoft, 2024-02-18:
@dsx724 What image does [libretech-slipstream](https://github.com/libre-computer-project/libretech-slipstream) customize?

guysoft, 2024-07-07:
Just updating I am progressing slowly on this because I have a daughter and a new job, but its still in the works.

guysoft, 2024-07-26:
The beta tag on docker and the branch are working now, most of the work is done. Just planning to add an automatic download type for latest rpi images.

guysoft, 2024-10-27:
This has been merged in to dev. I still need to update the docs.

guysoft, 2024-11-06:
Breaking change for github actions - make sure to add to your steps:
```
apt-get -y install python3-git
```

guysoft, 2024-11-24:
> Sorry I was sick. I opened a issue almost a decade ago but RPi doesn't care: [raspberrypi/linux#1761](https://github.com/raspberrypi/linux/issues/1761) Kind of why Libre Computer even exists :D Had enough hot garbage on the software and hardware sides.
> 
> @guysoft We have created this bugger: https://github.com/libre-computer-project/libretech-slipstream

Ok I had a look at @dsx724 It seems pretty similar in the idea to https://github.com/OctoPrint/CustoPiZer
You might take a look at that. 
Also CustomPiOS devel branch now works with Le Potato images. So you might like to try it out, I recommend the docker method. It brings a lot of tools you can already use.

guysoft, 2025-10-18:
Merged and there some time  closing

closed 2025-10-18 by guysoft