guysoft/CustomPiOS #216: Decided to improve the logo, and incorporate the new FullPageOS design into it

SecularSteve, 2024-02-08:
[image]
[image]

Couldn't decide if it looks better horizontal or vertical, so I made both for the vote.

Credit where it's due to [tailorvj](https://github.com/tailorvj) for his hand in designing the original. Full credit for the Raspberry in particular.

ZIP contains the SVGs and PNGs

[CustomPiOS_new.zip](https://github.com/guysoft/CustomPiOS/files/14205986/CustomPiOS_new.zip)

Noschvie, 2024-02-08:
For me the horizontal looks better, mayba with OS as horizontal too.

guysoft, 2024-02-11:
Hey, it looks cool, I need to think about it. I think the difference though is that the OS only reads in vertical, but the current logo is horizontal. If you could solve that it would help. It seems to me a horizontal logo would be better with what we are doing. L have a physical OctoPi and they also picked horizontal as how to print it.

SecularSteve, 2024-02-22:
Here's the horizontal version with the text at the bottom, as requested.

[image]
[image]

Once again, credit to [tailorvj](https://github.com/tailorvj) for part the design, and the ZIP contains the SVG and PNG.

[CustomPiOS_new3.zip](https://github.com/guysoft/CustomPiOS/files/14374539/CustomPiOS_new3.zip)

guysoft, 2024-06-10:
Just updating still undecided. Leaning towards it, but I have a few other things before changing the logo.