guysoft/CustomPiOS #218: Moving scripts from /home/pi/scripts to /opt/custompios/scripts

guysoft, 2024-03-23:
Since Rpi Foundation introduced a userless OS. Some hard-coded paths can now changed (for example in the gui module).
To fix this I am moving ``/home/pi/scripts to`` ``/opt/custompios/scripts``.
I am open to better ideas, but for now this keeps the best backwards compatability

guysoft, 2024-03-23:
@admin-toolkit you might want to update the ``admin-toolkit`` module
@asdf1nit you might want to update the ``cockpit-install`` module

guysoft, 2024-06-04:
No reply from @asdf1nit . evreyting else has been move, considering deprecating admin-toolkit and  cockpit-install if there is no maintainer.

asdf1nit, 2024-06-25:
@guysoft  sorry about the late reply. I will get the admin-toolkit module updated shortly. As for the cockpit-install, it can be deprecated as it no longer works in deb 12 buster; how should that be done?

guysoft, 2024-06-30:
@asdf1nit I think just deleting the folder and perhaps writing in the module docs "deprecated as it no longer works in deb 12 buster"