guysoft/CustomPiOS #220: The file `/boot/config.txt` has moved to `/boot/firmware/config.txt`

sunefred, 2024-06-16:
The "confix.txt" has moved from `/boot/config.txt` has moved to `/boot/firmware/config.txt` for bookworm based releases. The content of `/boot.config.txt` is now:

```
DO NOT EDIT THIS FILE

The file you are looking for has moved to /boot/firmware/config.txt
```

Other than the updated location, the content looks the same to me. This affects quite a few modules. A quick search through the repository found these:

- common.sh
- base
- gui
- raspicam
- usbconsole

I propose that these modules are updated such that the location of "config.txt" is stored in a variable that can be set or resolved based on the Raspberry PI OS release.

guysoft, 2024-06-16:
Thanks for reporting, will fix that.

sunefred, 2024-10-04:
**config.txt**

Here is an update regarding the new bookworm releases. You can still use and modify `boot/config.txt` during `chroot`. This will then get moved to `boot/firmware/config.txt` at some point during first boot.

**cmdline.txt**

Some settings have moved out of `config.txt` however. Raspberry PI OS (bookworm) now uses KMS (Kernel Model Setting) for display settings so in order to set screen resolution, for instance, you must modify the `cmdline.txt` file. During `chroot` this file is found in `/boot/cmdline.txt` and it is then moved to `/boot/firmware/cmdline.txt` during first boot.

https://www.raspberrypi.com/documentation/computers/configuration.html#set-the-kms-display-mode

closed 2025-11-27 by guysoft