guysoft/CustomPiOS #231: Problem with setup an image on MAC

EnvironCMO, 2024-10-10:
Hello,

I am currently having big problems setting up an image under MAC. I have done the following steps:

1. git clone https://github.com/guysoft/CustomPiOS.git
Feedback:
Clone to 'CustomPiOS'...
remote: Enumerating objects: 5148, done.
remote: Counting objects: 100% (1121/1121), done.
remote: Compressing objects: 100% (404/404), done.
remote: Total 5148 (delta 774), reused 862 (delta 685), pack-reused 4027 (from 1)
Receive objects: 100% (5148/5148), 1.03 MiB | 12.80 MiB/s, done.
Resolve differences: 100% (2773/2773), done.2nd command: ./src/make_custom_pi_os -g test
Feedback:
Settings:
making dstro in test
variant: raspios_lite_armhf
sed: 1: “config”: command c expects \ followed by text

After that I wanted to switch to Docker

An alternative way I also tried:
create a docker-compose.yml file in the src directory

version: '3.6'
services:
  custompios:
    image: guysoft/custompios:devel
    container_name: mydistro-builder
    privileged: true
    volumes:
      - ./:/os_make
    devices:
      - /dev/loop-control
    tty: true

-> docker-compose up -d

-> docker exec -it mydistro-builder CustomPiOS/make_custom_pi_os -g /os_make/ExampleDistro

Settings:
making dstro in /os_make/ExampleDistro
variant: raspios_lite_armhf
CustomPiOS/make_custom_pi_os: line 49: USER: unbound variable

Is this an error or is there a point by point guide to avoid running into this error

Translated with DeepL.com (free version)

unfixa1, 2024-10-11:
CustomPiOS/make_custom_pi_os: line 49: USER: unbound variable This error indicates that an undefined variable USER is used in the script. This may be because the environment variable is not set in the Docker container.

You can consider replacing docker with a tool such as [servbay](https://servbay.com), which is very suitable for your needs

guysoft, 2024-10-14:
@unfixa1 Is this an advertisement? Its certainly not what I would recommend, docker is the supported method here. Give me a reason no to delete your comment not to confuse people.

guysoft, 2024-10-14:
@EnvironCMO Your docker compose is missing the mounts to the distruction it should go in ``/distro``
Like this:

```
    volumes:
      - ./:/distro
```

You called your distribution there ``test`` not ``ExampleDistro``
(command I see that in is: ``command: ./src/make_custom_pi_os -g test``)

Here is a correct docker compose placed in your src folder if the "test" dir you wrote above:

```yaml
services:
  custompios:
    image: guysoft/custompios:devel
    container_name: test-build
    tty: true
    restart: always
    privileged: true
    volumes:
      - ./:/distro
    devices:
      - /dev/loop-control
    # dns:
    #   - 8.8.8.8
    #   - 4.4.4.4
```

Also note something about mac - dont run this in the ``/tmp`` folder. I just tried and found the mount didn't work because of that. make sure you are in your home directory or somewher that is not ``/tmp``

guysoft, 2024-10-27:
Closing if no reply

EnvironCMO, 2024-10-30:
sorry, i'm currently very busy with other topics. but i'll take a look at it.

sharpn, 2024-10-30:
Hey, I just followed this thread and seemed to get this working by setting 

```yml
 environment:
      - USER=root
```

in the docker file under custompios

full docker-compose as follows

```yml
services:
  custompios:
    image: guysoft/custompios:devel
    container_name: test-build
    tty: true
    restart: always
    privileged: true
    env_file: .env
    environment:
      - USER=root
    volumes:
      - ./:/distro
    devices:
      - /dev/loop-control
```

then running 

```bash
cd CustomPiOS
./make_custom_pi_os -g /distro/testing
```

from inside the docker container

guysoft, 2024-10-30:
@sharpn Strange, the user is already root. No need to set it. what error do you get without it?

sharpn, 2024-10-30:
it seems to be because this [line](https://github.com/guysoft/CustomPiOS/blob/devel/src/make_custom_pi_os#L49) is expecting an environment variable called `USER` it would probably fix it by changing it to `chown -R "(id -un")":"$(id -gn")" "${DEST}"`