guysoft/CustomPiOS #234: new release v1.5.0 breaks arm64 builds

khassel, 2024-10-26:
I set `BASE_ARCH=arm64` in variant config but this is not used anymore:

From the [logs](https://gitlab.com/khassel/magicmirroros/-/jobs/8194800582):

```bash
Sourcing variant config /distro/variants/arm64/config...
+++ set -a
+++ source /distro/variants/arm64/config
++++ BASE_ARCH=arm64
+++++ ls -t '/distro/image/*-raspbian-*-arm64-*.zip' '/distro/image/*-raspios-*-arm64-*.zip'
+++++ head -n 1
ls: cannot access '/distro/image/*-raspbian-*-arm64-*.zip': No such file or directory
ls: cannot access '/distro/image/*-raspios-*-arm64-*.zip': No such file or directory
++++ BASE_ZIP_IMG=
+++ set +a
+++ '[' -n '' ']'
+++ '[' -f /distro/config.local ']'
+++ TMP='base,magicmirroros, usage-statistics)'
+++ TMP='base,magicmirroros,usage-statistics)'
+++ MODULES_LIST=base,magicmirroros,usage-statistics,
+++ '[' -n /distro/workspace-arm64 ']'
+++ '[' -n /distro/workspace-arm64/mount ']'
+++ '[' -f /tmp/tmp.uUllhrq8SL ']'
+++ source /tmp/tmp.uUllhrq8SL
++++ export BASE_ARCH=armhf
++++ BASE_ARCH=armhf

...

+ echo 'Building on aarch64/arm64 device a armv7l system, using qemu-arm-static'
Building on aarch64/arm64 device a armv7l system, using qemu-arm-static
+ chroot . usr/bin/qemu-arm-static /bin/bash /chroot_script
qemu-arm-static: /bin/bash: Invalid ELF image for this architecture
++++ echo_red 'build failed, unmounting image...'
```

Looks like BASE_ARCH is overwritten again with `armhf`.

This worked before the new release.

khassel, 2024-10-26:
bug introduced with https://github.com/guysoft/CustomPiOS/commit/baaa155f33f54fa0e777039dd47cca751fa55802

guysoft, 2024-10-26:
I see the message:
```
+ echo 'No BASE_BOARD defined, defaulting to raspberrypiarmhf'
+ export BASE_BOARD=raspberrypiarmhf
```
It looks like V2 needs BASE_BOARD set.

I need to figure out of a way to not override the explicitly set settings. As a workaround you can set: 
```
build -b raspberrypiarm64
```
or for arm32
```
build -b raspberrypiarmhf
```
in here:
https://gitlab.com/khassel/magicmirroros/-/blob/master/build_img.sh?ref_type=heads#L27

You can also provide ``-d`` flag in build and it will download the correct image for you, that would simplify the script.

Another workaround would be to use the docker image in ``release/v1`` branch, it should work.

khassel, 2024-10-26:
thanks, will try your options, so I can replace my current ugly workaround overriding `custompios`:

```bash
      curl --output /CustomPiOS/custompios -L https://raw.githubusercontent.com/guysoft/CustomPiOS/c5b7ab895dde5b0fe43c1c8e55c8b86a3d5ced92/src/custompios
```

khassel, 2024-10-26:
Did not manage to get it running with `-d` option.

Fixed it now [with this commit](https://gitlab.com/khassel/magicmirroros/-/commit/552a33bef4f1a9900c1c2341d3d91d13407bf0c9), for me this is solved, feel free to close or leave open if you have still todo's ...

guysoft, 2024-10-27:
-b is --board and -d is --download.
perhaps you mixed them?

guysoft, 2024-11-06:
Fixed by https://github.com/guysoft/CustomPiOS/commit/c505f18ec0bde3fbe63f077328696070f8f48ad9 and https://github.com/guysoft/CustomPiOS/commit/b3f2c792e7fc5d4042916603e0ba42ab9316b52a
Just make sure to add
```
export BASE_BOARD=raspberrypiarm64
```

closed 2024-11-06 by guysoft