guysoft/CustomPiOS #236: Could not get image config

kraeml, 2024-10-29:
To show my problem I make the https://github.com/guysoft/CustomPiOS?tab=readme-ov-file#how-to-use-it with a dummy testPi.

Here is my history:

```bash
 2009  mkdir customKannWeg
 2010  cd customKannWeg
 2011  git clone https://github.com/guysoft/CustomPiOS.git
 2012  cd CustomPiOS/
 2013  ./src/make_custom_pi_os -g ../testPi
 2014  pyenv help global
 2015  pyenv global 3.12.3
 2016  ./src/make_custom_pi_os -g ../testPi
 2017  cd ../testPi/
 2018  vi src/config 
 2019  ls
 2020  cd src/
 2021  ls
 2022  cd ..
 2023  sudo src/build_dist 
 2024  cp ../CustomPiOS/src/images.yml src/
 2025  sudo src/build_dist 
```
I get this error massaging:

```
Distro path: /home/USER/Dokumente/tmp/customKannWeg/testPi/src
CustomPiOS path: /home/USER/Dokumente/tmp/customKannWeg/CustomPiOS/src
================================================================
+ /home/USER/Dokumente/tmp/customKannWeg/CustomPiOS/src/build ''
+ '[' -n '' ']'
+ LOG=build.log
+ define SCRIPT
+ IFS='\n'
+ read -r -d '' SCRIPT
+ true
+ '[' build.log '!=' no ']'
+ touch build.log
+ chmod 666 build.log
+ eval 'BUILD_SCRIPT_PATH=$(dirname $(realpath -s $BASH_SOURCE))
export EXTRA_BOARD_CONFIG=$(mktemp)
${BUILD_SCRIPT_PATH}/custompios_core/generate_board_config.py "${EXTRA_BOARD_CONFIG}"
echo "Temp source file: ${EXTRA_BOARD_CONFIG}"

source ${BUILD_SCRIPT_PATH}/common.sh
install_cleanup_trap

CUSTOM_OS_PATH=$(dirname $(realpath -s $0))

source ${CUSTOM_PI_OS_PATH}/config "${1}" "${EXTRA_BOARD_CONFIG}" ${@}
${CUSTOM_PI_OS_PATH}/config_sanity

[ "$CONFIG_ONLY" == "yes" ] || source ${CUSTOM_OS_PATH}/custompios ${@}
'
+ tee build.log
++++ realpath -s /home/USER/Dokumente/tmp/customKannWeg/CustomPiOS/src/build
+++ dirname /home/USER/Dokumente/tmp/customKannWeg/CustomPiOS/src/build
++ BUILD_SCRIPT_PATH=/home/USER/Dokumente/tmp/customKannWeg/CustomPiOS/src
+++ mktemp
++ export EXTRA_BOARD_CONFIG=/tmp/tmp.y47UYnWREB
++ EXTRA_BOARD_CONFIG=/tmp/tmp.y47UYnWREB
++ /home/USER/Dokumente/tmp/customKannWeg/CustomPiOS/src/custompios_core/generate_board_config.py /tmp/tmp.y47UYnWREB
Error: Could not get image config
+ exit 1
```

I can't find any information about image config and how to setup. I am using CustomPiOs in my project and have the same error.  Also in github runner fails with same error. Any idea what I can do.

IronWill79, 2024-11-01:
Hi. I've just done some experimentation with this and there needs to be an environment variable named `BASE_BOARD` that matches one of the branches of the `images.yml` file. I added `export BASE_BOARD="raspberrypiarmhf"` to my `build_dist` shell script before it calls `build_custom_os`

guysoft, 2024-11-03:
@IronWill79 Hey,
indeed BASE_BOARD is a new variable introduced on CustomPiOS v2 that just made it in to the devel branch.

In OPs code there is no image downloaded.
Ill have a look if any change that i missed that needs to be added to the make os scripts.

guysoft, 2024-11-06:
Found the issue, working on it. This error should be fixed, now, just making sure there are no other errors after it

guysoft, 2024-11-06:
Fixed by https://github.com/guysoft/CustomPiOS/commit/c505f18ec0bde3fbe63f077328696070f8f48ad9 and https://github.com/guysoft/CustomPiOS/commit/b3f2c792e7fc5d4042916603e0ba42ab9316b52a