guysoft/CustomPiOS #243: Variants Code in custompios doesn't match make_custom_pi_os

jxmx, 2025-01-31:
If I execute the following:

```bash
./src/make_custom_pi_os -v raspios_lite_arm64 -g ./asl3
cd asl3/src
sudo ./build_dist
```

Then `make_custom_pi_os` downloads the latest Raspbian lite image to a directory named `image`. However
when `custompios` is called out of `build_dist` it is looking for the file in a directory named `image-raspberrypiarm64`.
The script fails. Manually moving image to image-raspberrypiarm64 fixes the issue.

guysoft, 2025-02-02:
Hey, its because there is a v1 method and v2 method.
V1 used variants to build images, and v2 has a ``BASE_BOARD`` variable that should handle loading board-spesific settings

When using CustomPiOS v2 its recommended to use the docker build method, there is a new script:
https://github.com/guysoft/CustomPiOS/blob/devel/src/custompios_core/base_image_downloader.py
In docker its run using the ``--download`` or ``-d`` flag. You can see it here:
https://github.com/guysoft/CustomPiOS/wiki/Building-with-Docker#docker-run

I need to see how your specific flow works because I tested it mostly on the docker build method - is there a reason you are not using it? It tends to be better in handling that.
I think running ``base_image_downloader.py`` with the right env would download the right images but I have yet to test it. You are welcome to try and report. I am pretty swamped these days.

jxmx, 2025-02-03:
The flow is basically what I put above. It's being run via a GitHub Runner on a purpose-started AWS Gravitron instance. Docker on top of that seemed unnecessary. I'll have to look closer.

closed 2025-02-03 by jxmx