guysoft/CustomPiOS #252: docker image not available anymore

khassel, 2025-03-15:
The docker image is missing on [docker hub](https://hub.docker.com/r/guysoft/custompios/).

Hopefully not intended ...

guysoft, 2025-03-15:
I didnt do anything - perhaps no one is pulling it?
you changed the location right?

khassel, 2025-03-15:
> you changed the location right?

don't know what you are talking about here ...

I use the image weekly for mmos builds, now I get [this](https://gitlab.com/khassel/magicmirroros/-/jobs/9421014770).

```bash
❯ docker pull guysoft/custompios:devel
Error response from daemon: pull access denied for guysoft/custompios, repository does not exist or may require 'docker login'
```

If I follow the link to docker hub posted in the description I'm getting a 404.

guysoft, 2025-03-16:
Ah you mean CustomPiOS.
This is really strange!
The repo vanished from docker hub, and there is no trace or email from them.
Before I re-create it. I have sent an email to them. If this is how dockerhub looks after images I am considering moving to another docker registry if you have any recommendations.

guysoft, 2025-03-17:
Ok, so I actually got a reply from DockerHub.

They say that the image was reported for is collecting and sending system information to a third party without disclosure, which violates our terms of service 3.2.

I am not aware of the CustomPiOS doing something like this and the CI/CD of the image is open and visible in this repo.
So this is really strange. The only thing I can think of that is remotely connected to this is the users-statistics module. Which someone reported last week here:
https://github.com/guysoft/HotSpotOS/issues/9 .
Note this is an optional module so its opt-in for CustomPiOS developers.
I could re-create the image and add a disclosure about the module. That should not violate the term of service.
However it does not stop random people reporting it and closing it down.
@ztnewman if you have any insights, no hard feelings - but I would like to get the image up online because its breaking other people's images too.

Moreover there is an identical test repo with the same code that is still up. 

I am considering opening a new registry to serve CustomPiOS since they obviously just took it down without investigating the intentions or cause of the report.

guysoft, 2025-03-17:
Well I have to change the container because I can't create the same name. So might as well try github container registry
I get this:

[image]

guysoft, 2025-03-17:
WIll work on creating
```
ghcr.io/guysoft/custompios:devel
```

guysoft, 2025-03-18:
Ok, moved to ``ghcr.io/guysoft/custompios:devel``

Please update your CI/CD docker-compose.yml etc.

Sorry I have no way to migrate it smoothly

closed 2025-03-18 by guysoft

khassel, 2025-03-18:
> Please update your CI/CD docker-compose.yml etc.
> Sorry I have no way to migrate it smoothly

no problem, thanks for fixing ...

guysoft, 2025-03-23:
```
Hello from Docker Support,

We have reviewed with our security team and the referenced "guysoft/custompios" repository has been restored to your account on Docker Hub.

Regards,
Docker Support
```

Since i set it up, ill push both to dockerhub and github packages once i update the yaml