guysoft/CustomPiOS #253: autologin-session=guisession

iwl, 2025-03-18:
https://github.com/guysoft/CustomPiOS/blob/1027bcbd3a19a4cb6f923b3ed9874aaa70775eac/src/modules/gui/end_chroot_script#L11

`autologin-session=guisession` - - - - - should also be set
`sed -i 's/#autologin-session=.*/autologin-session=guisession/g' /etc/lightdm/lightdm.conf`
I think that ensures always guisession is used on autologin.
Otherwise it somehow depends on last session used and I have seen some hacks with ~/.dmrc, but still have seen the Desktop sometimes after boot.
I have also set XKBOPTIONS="terminate:ctrl_alt_bksp":
/etc/default/keyboard:
`#KEYBOARD CONFIGURATION FILE

#Consult the keyboard(5) manual page.

XKBMODEL="pc105"
XKBLAYOUT="de"
XKBVARIANT=""
XKBOPTIONS="terminate:ctrl_alt_bksp"

BACKSPACE="guess"`
terminate:ctrl_alt_bksp -> I can kill X-Server with ctrl_alt_bksp and get to X-Server-Login, where I can login for the GUI and do some maintenance, or can press ctrl_alt_bksp again and get Autologin FullPage again,
so  ctrl_alt_bksp Alternates between X-Login and Autologin FullPage

guysoft, 2025-03-18:
Ok, it worked for me until now - does it work for you and this is just a precaution or did this break something for you?
I want to understand the urgency.

iwl, 2025-03-20:
I've experienced AutoLogins into the LightDmDesktop.
That's why I investigated the Problem.
And I think that ~/.dmrc writing in some scripts of the image I used was somebodies try to solve the same problem, but haven't worked dependable for me.
Several can't cope with a Linux Desktop appearing, fewer can access the System to easy then...

guysoft, 2025-03-23:
Ok - any chance you can draft a PR? I will get to it far later then you likely.