guysoft/CustomPiOS #256: `systemctl_if_exists` function in src/common.sh does not fail gracefully if the service does not exist

platima, 2025-05-04:
In building MainsailOS I found that the line `systemctl_if_exists disable armbian-firstrun-config.service` caused the entire build to fail because that service did not exist.

I CBF with a PR but this expansion on the function should help:

`bashfunction systemctl_if_exists() {
    if ! hash systemctl 2>/dev/null; then
        echo_red "no systemctl, not running"
        return 0
    fi
    
    # Check if the service exists before running the command
    local service=""
    for arg in "$@"; do
        if [[ "$arg" == *.service ]]; then
            service="$arg"
            break
        fi
    done
    
    if [ -n "$service" ] && ! systemctl list-unit-files "$service" &>/dev/null; then
        echo_red "$service does not exist, skipping"
        return 0
    fi
    
    # If we got here, both systemctl exists and the service exists
    systemctl "$@"
    return $?
}`

guysoft, 2025-05-04:
Can you post the actual error from your log? I don't understand in what way and where it failed

platima, 2025-05-04:
It was from MainsailOS build, and gave "failed to disable unit, unit armbian-firstrun-config.service does not exist." due to the base image not having that service.

As I'm sure you can discern, this change just checks that the service exists before taking any action on it, so that it can fail softly and not fail the whole build. There are rare edge cases where this would fail of course, but significantly less than the original implementation.

Take it or leave it, I am not particularly bothered. I just fixed up this code for myself and everything is running happily now so I thought I'd pass it upstream to help improve your code, as is the spirit of open source.

Best of luck either way, and thanks for CustomPiOS.

platima, 2025-05-04:
Mate I wrote it in vim and it works fine for me. As I said, there are cases where it fails and I was just trying to share what I did to fix an issue with an edge case. It could also work for me and not you, as I'm using it to build MainsailOS, not a stock CustomPiOS.

If you want to be an a-hole about it you're very welcome to, but how about just closing the PR and moving on with your life. Not everyone needs to be as sad as you seem to be.

platima, 2025-05-04:
> It looks like there is a loop that runs but then breaks - then uses only one argument - did you actually test this or just used AI blindly? This will fail on if there are multiple args

Yep did exactly that smart guy. How could you tell, was it because I tried to help or my code was different to yours? 

Solve it yourself, as I said - I don't particularly care and was trying to help, but have no interest in dealing with people that are just miserable.

closed 2025-05-04 by platima

guysoft, 2025-05-04:
@platima Please be kind, we follow here the OctoPrint code of conduct.

https://octoprint.org/conduct/

platima, 2025-05-04:
Mate I was just following your tone, as I found your remark to be rather facetious. It could be worth you reading that link yourself to be honest.

As I said, I tried to help, and I told you it may not be perfect. You're welcome to do with it what you want, I'm unsubscribing from this thread now as I have no interest in participating with someone like yourself.

Thanks again for the project, even though I do not use it directly, and best of luck.